libretro/Lakka-LibreELEC #850: RPi4 Bluetooth Keyboard Re-connect Issue

kmssd, 2019-11-21:
- Which version of Lakka are you using? 2.3.1
- What system hardware are you using? RPi4 (4gb)
- What did you do? Pair and connect bluetooth keyboard for use (now and after reboots/shutdowns)
- What did you expect to happen? Keyboard to work after start-up
- What happened instead? I have to toggle the keyboard on and off mutliple times before it reconnects

```
[INFO] RetroArch 1.7.8 (Git ea7e682253)
[INFO] === Build =======================================
[INFO] Capabilities: NEON VFPv3 VFPv4
[INFO] Built: Sep 16 2019
[INFO] Version: 1.7.8
[INFO] Git: ea7e682253
[INFO] =================================================
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Redirecting save file to "/storage/savefiles/.srm".
[INFO] Redirecting savestate to "/storage/savestates/.state".
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 30000.00 Hz.
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[WARN] [DRM]: Couldn't get device resources.
[WARN] [DRM]: Couldn't get device resources.
[INFO] [DRM]: Found 1 connectors.
[INFO] [DRM]: Connector 0 connected: yes
[INFO] [DRM]: Connector 0 has 37 modes.
[INFO] [DRM]: Connector 0 assigned to monitor index: #1.
[INFO] [DRM]: Mode 0: (1920x1080) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 1: (1920x1080) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 2: (1920x1080) 1920 x 1080, 50 Hz
[INFO] [DRM]: Mode 3: (1920x1080) 1920 x 1080, 30 Hz
[INFO] [DRM]: Mode 4: (1920x1080) 1920 x 1080, 30 Hz
[INFO] [DRM]: Mode 5: (1920x1080) 1920 x 1080, 25 Hz
[INFO] [DRM]: Mode 6: (1920x1080) 1920 x 1080, 24 Hz
[INFO] [DRM]: Mode 7: (1920x1080) 1920 x 1080, 24 Hz
[INFO] [DRM]: Mode 8: (1600x1200) 1600 x 1200, 60 Hz
[INFO] [DRM]: Mode 9: (1680x1050) 1680 x 1050, 60 Hz
[INFO] [DRM]: Mode 10: (1280x1024) 1280 x 1024, 75 Hz
[INFO] [DRM]: Mode 11: (1280x1024) 1280 x 1024, 60 Hz
[INFO] [DRM]: Mode 12: (1440x900) 1440 x 900, 75 Hz
[INFO] [DRM]: Mode 13: (1440x900) 1440 x 900, 60 Hz
[INFO] [DRM]: Mode 14: (1280x960) 1280 x 960, 60 Hz
[INFO] [DRM]: Mode 15: (1280x800) 1280 x 800, 60 Hz
[INFO] [DRM]: Mode 16: (1152x864) 1152 x 864, 75 Hz
[INFO] [DRM]: Mode 17: (1280x720) 1280 x 720, 60 Hz
[INFO] [DRM]: Mode 18: (1280x720) 1280 x 720, 60 Hz
[INFO] [DRM]: Mode 19: (1280x720) 1280 x 720, 50 Hz
[INFO] [DRM]: Mode 20: (1024x768) 1024 x 768, 75 Hz
[INFO] [DRM]: Mode 21: (1024x768) 1024 x 768, 70 Hz
[INFO] [DRM]: Mode 22: (1024x768) 1024 x 768, 60 Hz
[INFO] [DRM]: Mode 23: (832x624) 832 x 624, 75 Hz
[INFO] [DRM]: Mode 24: (800x600) 800 x 600, 75 Hz
[INFO] [DRM]: Mode 25: (800x600) 800 x 600, 72 Hz
[INFO] [DRM]: Mode 26: (800x600) 800 x 600, 60 Hz
[INFO] [DRM]: Mode 27: (800x600) 800 x 600, 56 Hz
[INFO] [DRM]: Mode 28: (720x576) 720 x 576, 50 Hz
[INFO] [DRM]: Mode 29: (720x480) 720 x 480, 60 Hz
[INFO] [DRM]: Mode 30: (720x480) 720 x 480, 60 Hz
[INFO] [DRM]: Mode 31: (640x480) 640 x 480, 75 Hz
[INFO] [DRM]: Mode 32: (640x480) 640 x 480, 73 Hz
[INFO] [DRM]: Mode 33: (640x480) 640 x 480, 67 Hz
[INFO] [DRM]: Mode 34: (640x480) 640 x 480, 60 Hz
[INFO] [DRM]: Mode 35: (640x480) 640 x 480, 60 Hz
[INFO] [DRM]: Mode 36: (720x400) 720 x 400, 70 Hz
[INFO] [GL]: Found GL context: kms
[INFO] [GL]: Detecting screen resolution 1920x1080.
[INFO] [EGL] Found EGL_EXT_platform_base, trying eglGetPlatformDisplayEXT
[INFO] [EGL]: EGL version: 1.4
[INFO] [EGL]: Current context: 0xb4007fc0.
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
[INFO] [GL]: Vendor: Broadcom, Renderer: V3D 4.2.
[INFO] [GL]: Version: OpenGL ES 3.0 Mesa 19.1.1.
[INFO] [GL]: Using resolution 1920x1080
[INFO] [GL]: Default shader backend found: glsl.
[INFO] [Shader driver]: Using GLSL shader backend.
[WARN] [GL]: Stock GLSL shaders will be used.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] Setting up menu pipeline shaders for XMB ...
[INFO] [GLSL]: Compiling ribbon shader..
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Compiling simple ribbon shader..
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Compiling snow shader..
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Compiling modern snow shader..
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Compiling bokeh shader..
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Compiling snowflake shader..
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GL]: Using 4 textures.
[INFO] [GL]: Loaded 1 program(s).
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Video]: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
[INFO] [udev]: Keyboard #0 (/dev/input/event0).
[INFO] [udev]: Mouse #0 (/dev/input/event3).
[INFO] [udev]: Mouse #1 (/dev/input/mouse0).
[INFO] [Joypad]: Found joypad driver: "udev".
[WARN] [udev]: Full-screen pointer won't be available.
[INFO] [Video]: Found display server: null
[INFO] Found shader "/tmp/shaders/bilinear.glslp"
[INFO] Found shader "/tmp/shaders/nearest.glslp"
[INFO] Found shader "/tmp/shaders/slang-conversion-template.glsl"
[INFO] Found shader "/tmp/shaders/stock.glsl"
[INFO] ALSA: Using signed 16-bit format.
[INFO] ALSA: Period size: 444 frames
[INFO] ALSA: Buffer size: 1411 frames
[INFO] [Menu]: Found menu display driver: "gl".
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Menu]: Found menu display driver: "gl".
[CDROM] No sg devices found and sg kernel module is not loaded.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[WARN] [XMB] Asset missing: /tmp/assets/xmb/flatui/png/disc.png
[INFO] [LED]: LED driver = 'null' 0x35f134
[INFO] [MIDI]: Initializing ...
ALSA lib /home/natinusala/release/Lakka-LibreELEC/build.Lakka-RPi4.arm/alsa-lib-1.1.9/src/seq/seq_hw.c:466:(snd_seq_hw_open) open /dev/snd/seq failed: No such file or directory
[ERROR] [MIDI]: snd_seq_open failed with error -2.
[ERROR] [MIDI]: Initialization failed (list of input devices unavailable).
[ERROR] Implementation uses threaded audio. Cannot use rewind..
[INFO] SRAM will not be saved.
[INFO] Loading history file: [/storage/.config/retroarch/content_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_music_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_video_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_image_history.lpl].
[INFO] Loading favorites file: [/storage/.config/retroarch/content_favorites.lpl].
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
```

zeno-mcdohl, 2020-03-14:
I am also experiencing this with any Bluetooth controller on the RPi4. I have verified the BT service boots up at startup, yet still does not re-pair.

kmssd, 2020-03-14:
On an older verison of Lakka it worked fine. Maybe they updated the Bluetooth service?

closed 2020-08-18 by Ntemis

zeno-mcdohl, 2020-08-19:
Is this resolved?

Ntemis, 2020-08-19:
2.3.x is eol try a fresh nightly image and open a new issue if anything comes up. Thank you in advance

kmssd, 2020-08-19:
> 
> 
> Is this resolved?

No, it was never fixed.

Ntemis, 2020-08-19:
Thank you for your contribution but unfortunately there wont be any new versions released from this branch as it has being decommissioned.
Your interest in transforming Lakka OS into a better distro is highly appreciated.
All efforts are now moved on for next major release and we continue passionate and focused to deliver the best RetroArch Turn-key Distro OS.
Feel welcome to test and report any issues by testing the new nightly images from here https://nightly.builds.lakka.tv/