libretro/Lakka-LibreELEC #851: Lakka on RPi4 fails to output dpi24 video

kevinfishburne, 2019-11-26:
Running official Raspbian (2019-09-26-raspbian-buster.img) with attached config files properly outputs dpi24 video over GPIO to HATs, yet Lakka nightly (Lakka-RPi2.RPi4.arm-nightly-2019-11-04.img) fails with the same config files.

What is the difference between Raspbian's and Lakka's dpi24 overlay (or other part of graphics stack) with respect to dpi24 output on the RPi4? This *may* be due to a difference in firmware between Lakka and Raspbian (if there is one).

[240pi.txt](https://github.com/libretro/Lakka-LibreELEC/files/3890938/240pi.txt)
[config.txt](https://github.com/libretro/Lakka-LibreELEC/files/3890939/config.txt)

natinusala, 2019-11-27:
Our overlay might be outdated, do you know when it was added to Raspbian?

kevinfishburne, 2019-11-28:
For the 2019-09-26 Raspbian release the dpi24.dtbo file is dated 2019-08-19 in the image and is 975 bytes.

Anyone have a RPi4 and dpi24 display they can put Lakka on to test using dpi_timings (formerly hdmi_timings)? Despite a lot of experimenting with settings and copying files from Raspbian the best I've got is getting Lakka to display the logo/berry/flower. It appears to be loading and running, but acts like the dpi24/GPIO display is a dummy TTY or something.

kevinfishburne, 2019-12-16:
I somehow got it working using the Nov. 4th nightly. Here are my config files.
[240pi.txt](https://github.com/libretro/Lakka-LibreELEC/files/3966136/240pi.txt)
[config.txt](https://github.com/libretro/Lakka-LibreELEC/files/3966137/config.txt)
The RPi 4 also now works with a super-resolution of 3840x240.

closed 2019-12-16 by kevinfishburne