libretro/Lakka-LibreELEC #852: Feature request: Manually edit playlists via UI

desperateCoder, 2019-12-01:
Hey guys!

As a developer i'm familiar to JSON and text-based data structures, but my best friend and gaming enthusiast isn't.

Editing the playlist files or generating them via tools isn't impossible, but i think the majority of the people would love an easy way to edit the playlists via the lakka UI.

We can already remove games from the playlists, but i didn't find a way to manually add an unrecognized game (not in DB) to a specific tab other than "Favourites", nor add a new gaming platform tab.

What are your thoughts about that?

natinusala, 2019-12-01:
Hello,

This is a good idea, you should submit an issue on RetroArch to ask for this feature :)

RobLoach, 2019-12-02:
You can enable "Rename" and "Remove" in Settings --> Playlists. There's also some work being done on getting manual scans in place to add custom playlists manually.

desperateCoder, 2019-12-02:
@natinusala done.

@RobLoach good to know, there is some work in progress, hope this feature comes soon. Maybe our friends at RetroArch will make it happen soon :smiley:

desperateCoder, 2019-12-02:
Well, they did it. Yesterday. Just as I opened this issue. \*Shrug\*

Any plans, when this will be available for lakka users? 😅

natinusala, 2019-12-02:
When we'll update Lakka with RA 1.8.3 or newer :)

desperateCoder, 2019-12-02:
@natinusala I'm looking forward to the release! Thank you guys too for your awesome work! 

Feel free to close this issue, as soon as it's released, so I can instantly upgrade my best friends lakka instance. And my girlfriend's. And her colleague's. Lakka ftw!!! 😄

closed 2020-08-18 by Ntemis