libretro/Lakka-LibreELEC #853: Ocarina of Time Seg fault

Mikau28, 2019-12-01:
If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
Stable 2.3.1
- What system hardware are you using?
x64 Generic// Lenovo m93p
- What did you do?
Started Ocarina of Time, pressed start in overworld.
- What did you expect to happen?
Subsystem menu shoiuld open
- What happened instead?
Seg fault

[log.txt](https://github.com/libretro/Lakka-LibreELEC/files/3908808/log.txt)

gouchi, 2019-12-05:
Hi,

Do you reproduce the issue with the [nightly build](http://nightly.builds.lakka.tv/2019-11-04/Generic.x86_64/Lakka-Generic.x86_64-nightly-2019-11-04.img.gz) ?

Thank you.

Mikau28, 2019-12-07:
Yes the same issue happens on LE 9.2 build

gouchi, 2019-12-07:
Thank you for the feedback may you try also with [this build](http://nightly.builds.lakka.tv/PRs/pr_838_000118/Generic.x86_64/).

Thank you.

Mikau28, 2019-12-08:
> Thank you for the feedback may you try also with [this build](http://nightly.builds.lakka.tv/PRs/pr_838_000118/Generic.x86_64/).
> 
> Thank you.

Issue resolved with this pr build.

closed 2019-12-08 by Mikau28