libretro/Lakka-LibreELEC #857: Graphical Artifacting on Main Menu

MicrosoftTakeover, 2019-12-13:
- Which version of Lakka are you using?
Lakka-RPi.arm-2.3.1.img
- What system hardware are you using?
Raspberry Pi Model B
- What happened?
Scrolling down large lists eventually causes graphical artifacts, first the boxarts turn black and then eventually the whole interface starts disappearing. This is pretty hard to replicate, sometimes I have to scroll through thousands of menu items before it happens.

natinusala, 2019-12-13:
This is a known issue, however it should have been fixed... I wasn't expecting it to come back

gouchi, 2019-12-17:
Surely related to this RA [issue](https://github.com/libretro/RetroArch/issues/6747) and same [issue](https://github.com/libretro/Lakka-LibreELEC/issues/541) on Lakka.

Same as [this video](https://www.youtube.com/watch?v=0to3Kznk3Z4) ?

MicrosoftTakeover, 2019-12-18:
Thanks for the video, yup - it's the same issue.

closed 2020-08-18 by Ntemis