libretro/Lakka-LibreELEC #861: Black screen after boot from USB on Macbook Pro 13,1.

smitheric95, 2019-12-30:
> Which version of Lakka are you using?

I've tried version `Lakka-Generic.x86_64-2.3.1.img` and `Lakka-Generic.x86_64-nightly-2019-11-04.img`

> What system hardware are you using?

[Macbook Pro 13,1](https://everymac.com/systems/apple/macbook_pro/specs/macbook-pro-core-i7-2.4-13-late-2016-retina-display-no-touch-bar-specs.html) with Intel i7 (6660U) and Iris Graphics 540 running MacOS Catalina 10.15.1. Notice, this model does NOT have Apple's new T2 chip. 

> What did you do?

1. Flashed the img to a thumb drive using Etcher as [described](http://www.lakka.tv/get/macos/generic/install/).
2. Booted into EFI mode by holding `Option` and selecting the thumb drive. 

> What did you expect to happen?

For the Lakka flower to appear and the thumb drive initialization to begin. This is what happened when I used the same thumb drive (after reflashing) on my [iMac](https://everymac.com/systems/apple/imac/specs/imac-core-i5-2.7-21-inch-aluminum-late-2012-specs.html) with NVIDIA GeForce GT 640M. After the reboot, I was able to boot into the live version of Lakka just fine. 

> What happened instead?

All I see after selecting EFI Boot is a black screen, the Lakka logo never shows up.

---

Some debugging steps I took: 

1. Verified I could boot from the same USB thumb drive by flashing a Kubuntu img on it. (This worked fine on my Macbook.)
2. Plugged an external keyboard and monitor to the MacBook to try to circumvent any built-in keyboard/display issues. 
3. Tried multiple pre-built releases as mentioned above. 
4. Messed around with the some of the parameters under the `live` option in `syslinux.cfg`. Was hoping to get some error messages but was unsuccessful. 

Unfortunately, I don't have logs to share as I don't make it very far in the boot process. :(

nunayabizness, 2020-01-31:
OpenFirmware on Macbooks doesn't properly provision the PCI-E bus master for NVIDIA drivers, so it has to be set by the bootloader or drivers.  The error doesn't occur for Nouveau drivers (which is why you can EFI install ubuntu and other distros and get a working system, but distros such as Lakka which include the nvidia drivers break), so it can be assumed that Nouveau automatically provisions while the NVIDIA drivers aren't "smart" enough to do it on their own.
All you have to do (on most systems) is modify grub to provision the bus master and enable vga support.  Use `lshw` to find the bus identifiers then place them in /etc/grub.d/01_enable_vga.conf.

Quoting:

> 2. Find the right PCI-E bus identifiers
>
>We need both the IDs for the graphics card and the PCI-E bridge that it is connected to. Issue the following command in a shell:
>
>~$ sudo lshw -businfo -class bridge -class display
>pci@0000:00:00.0              bridge         MCP89 HOST Bridge
>pci@0000:00:03.0              bridge         MCP89 LPC Bridge
>pci@0000:00:0e.0              bridge         NVIDIA Corporation
>pci@0000:00:15.0              bridge         NVIDIA Corporation
>pci@0000:00:16.0              bridge         NVIDIA Corporation
>pci@0000:00:17.0     >!!<     bridge         MCP89 PCI Express Bridge
>pci@0000:04:00.0     >!!<     display        MCP89 GeForce 320M
>
>Have a look at (1) the line saying display and (2) the line with bridge right before that display line. >Write down the PCI-E bus ids (format XX:YY.Z) of the bridge device (here 00:17.0) and the display device (here 04:00.0) and remember which is which. Note: Those IDs may be different on your machine, depending on your Mac model and revision.
>3. Create a GRUB script for setting the PCI-E registers during boot
>
>Fire up a text editor with sudo nano /etc/grub.d/01_enable_vga.conf and copy/paste the content below. Replace 00:17.0 with the PCI-E ID of your bridge device noted in step 2. Replace 04:00.0 with the PCI-E ID of your display device noted in step 2.
>
>cat << EOF
>setpci -s "00:17.0" 3e.b=8
>setpci -s "04:00.0" 04.b=7
>EOF
>
>Finally, make the created file executable and update your grub config files using the following TWO commands.
>
>~$ sudo chmod 755 /etc/grub.d/01_enable_vga.conf
>~$ sudo update-grub
[https://askubuntu.com/questions/264247/proprietary-nvidia-drivers-with-efi-on-mac-to-prevent-overheating](url)

smitheric95, 2020-01-31:
@nunayabizness 
I don't actually have an NVIDIA card on my MacBook, I have an integrated one. 

I assume the same logic applies though?

nunayabizness, 2020-02-01:
Very much.  I somehow misattributed the 640M on your imac to your macbook pro when I read your post, but my guess is that there is the same or a very similar bug has occurred with the iris graphics.

A quick search shows that there are problems with enumeration/provisioning with your model year macbook pro, so my guess is likely not too far off the mark. [https://www.macrumors.com/2016/11/18/macbook-pro-intel-iris-pro-580-misreporting/](url)

closed 2020-08-18 by Ntemis