libretro/Lakka-LibreELEC #863: [Feature Request] Button to "Factory reset"

coldspark29, 2019-12-30:
It would also be nice to have a button to factory reset Lakka. There are cases, when you forget the settings you have made and just want to return to the initial state.

natinusala, 2019-12-31:
This is a good idea, it can be achieved by formatting the storage partition using a PC. If you only want to reset your settings, remove .config/retroarch

coldspark29, 2020-01-10:
I already did it manually, but it is a hassle. LibreELEC has a button, where it returns the installation to the initial state, which comes in handy sometimes. My problem was that my SD card was out of memory and that broke LAKKA somehow. I guess some bytes were overwritten or something. Had to reinstall it all and with NOOBS it means, that you have to do everything from scratch.

natinusala, 2020-01-10:
I wonder how LibreELEC implements that, I suppose we can do the same on the LE9 branch

coldspark29, 2020-01-10:
Asking is always for free :)

ToKe79, 2020-01-15:
I think we can do it using a systemd unit or script at startup, which would look for /storage/.factory_reset and then remove /storage/.config/retroarch folder. This must be done when RetroArch is not running.

ToKe79, 2020-01-15:
In RetroArch we just need to add a menu-entry, which would create that blank file and inform the user that they should reboot their device.

natinusala, 2020-01-15:
To me a factory reset would be more like formatting the storage partition entirely, like what Android does

closed 2020-08-18 by Ntemis