libretro/Lakka-LibreELEC #864: No Wifi due to missing firmware for Realtek RTL8188EUS.

mxm-tr, 2019-12-30:
Hi everyone, I'm trying to get my Wifi dongle to work but I think a firmware is missing:
I'm running RPi2.arm-2.3.1 on Raspberry 2B+

After plugging the TP-LINK wifi adapter, and got the following:
`#lsusb
Bus 001 Device 004: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter

I expected to get a new network interface but no luck so far.

Here's what I noticed:
` # ls /lib/firmware/rtlwifi/rtl8188e*
/lib/firmware/rtlwifi/rtl8188efw.bin   /lib/firmware/rtlwifi/rtl8188eufw.bin
`
So I guess a /lib/firmware/rtlwifi/rtl8188eus.bin is missing? Or is there anything else that is been done wrong?

Not having gcc to build the driver myself is kind of anoying, so posting here is my last option. Thanks for your hard work guys!

Also, dmesg:

`[    1.899041] usb 1-1.4: new high-speed USB device number 4 using dwc_otg
[    1.989697] usb 1-1.4: New USB device found, idVendor=0bda, idProduct=8179
[    1.989709] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    1.989718] usb 1-1.4: Product: 802.11n NIC
[    1.989726] usb 1-1.4: Manufacturer: Realtek
[    1.989735] usb 1-1.4: SerialNumber: 00E04C0001
[   22.096956] EXT4-fs (mmcblk0p2): couldn't mount as ext3 due to feature incompatibilities
[   22.097659] EXT4-fs (mmcblk0p2): couldn't mount as ext2 due to feature incompatibilities
[   22.173732] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)
`

ToKe79, 2019-12-31:
Hi,
I think the latest nightly should include the firmware, this was fixed with 4d00e847edf0a585f2fa7498be0c8bb282117b11.
Latest nightly for RPi2: http://nightly.builds.lakka.tv/2019-11-04/RPi2.arm/
Please, let us know if the issue is fixed using the above image. Thank you.

mxm-tr, 2020-01-06:
Thanks, using the nightly worked just fine!

closed 2020-01-07 by ToKe79