libretro/Lakka-LibreELEC #869: Playlist Thumbnail errors

coldspark29, 2019-12-31:
- Which version of Lakka are you using? 2.3.1
- What system hardware are you using? RPi4
- What did you do? Load core packages from "Download content"
- What did you expect to happen? No false thumbnails to show
- What happened instead? Sometimes the thumbnails are from other games and cores even. They also load very slowly. The problem gets worse, while updating the playlist/scanning directories.

natinusala, 2020-01-02:
This issue might already be fixed in newer RetroArch, although that seems to be a pretty serious one. I will link new images when the next libretro update is done, so that you can see if the error has been fixed

coldspark29, 2020-01-10:
I hope so. It is not dramatic though. Seems to only happen during playlist update. Afterwads everything is fine.

closed 2020-08-18 by Ntemis