libretro/Lakka-LibreELEC #870: Lakka/RPI4- no bluetooth and no ssh connection

saralabadie, 2020-01-04:
Hello,
I'm a noobs and I know this may not be the right place but I've tried looking online everywhere.. I've bought a RPI4 and I'm using the absolute latest version of Lakka from the nightly builds (11-04-19). 

I cannot connect ANY bluetooth no matter what I do. It's on in services and my wifi is on. And then when I've tried to access the ssh on my Mac terminal, whenever I enter my IP address (triple checked on my computer and Lakka's information that it's right)
**ssh root @IP#** .. the terminal doesn't let me enter a password. It sticks and I can't type anything and therefore cannot connect to Lakka with the ssh. I'm afraid I typed something with password that has now frozen it, made it public, etc. I don't know.. is there code I can type in to reset a password. I know the default is root but I can't type anything.

My two controllers are 8Bitdo SF30 nintendo Gamepad and the 8bitdo Pro. Only the SF30 works while plugged in using the USB but both won't work together plugged in - the Pro gamepad makes the menu go crazy and neither are recognized within Lakka (I can only see the 8Bitdo SF30 device when plugged in) 

So i'm hoping someone can help me - this may be an issue for reddit but I found this useful forum and thought I'd try here hoping someone would be willing to help a newbie!
Thank you kindly.

leeomara, 2020-01-10:
It may be a typo in your comment, but in case not: there should be no space between `root` and the `@ip` parts.

For example, if your RPi4 is at the IP address 192.168.10.22, the `ssh` command would be:

    ssh root@192.168.10.22

coldspark29, 2020-01-12:
You can also use 

`root@lakka`

I have absolutely no problems with my RPi4. I am on Linux though. It might be your Mac or the issue occurs due to the wrong ECDSA host key. I have to reenter it in ~/.ssh/known_hosts almost every time, because I also have Kodi and Raspbian on the machine. All three have different keys and ssh will not prompt you for the password, if you don't change it to the system you are using currently. But you should get a message about that, I think. No idea how it is on Macs.

So please paste all the error messages you get! Otherwise we can only guess

closed 2020-08-18 by Ntemis