libretro/Lakka-LibreELEC #871: Lakka is very slow on 4k TV

Pxtl, 2020-01-06:
> If you are sure that it's a bug in Lakka, please answer the following questions:
> - Which version of Lakka are you using?

Lakka 2.3.1 - I have not tested this in the nightly because the datestamps in http://le.builds.lakka.tv/RPi2.RPi4.arm/ are from about the same date as 2.3.1.  Are nightlies no longer a thing?

> - What system hardware are you using?

Pi4 2GB

> - What did you do?

Plugged my Lakka Pi4 into a Samsung 4K TV

> - What did you expect to happen?

Run at the same speed it ran on my normal Toshiba 1080P TV.

> - What happened instead?

Slow emulation, frequent freezes.

Workaround was to force Lakka into 1920x1080 resolution.  This fixed the problem.

Ideally, Lakka would have a "max auto resolution" config and images for low-power devices like Allwinner and Pi would prevent auto-resolutions above 1920x1080.

Ntemis, 2020-01-06:
if we hardcode the resolution then it wont display anything on e.g. HD ready TV's(720p) and i dont think we want this to happen or we force all rpi4 users to get FHD tv's as a minimum and by it self is unaccepted behavior from its genesis

Pxtl, 2020-01-07:
@ntemis, I meant introducing a new "max resolution" feature for lower-power device images.  Like on a Pi, don't try to go above 1080p.  If you plug it into a 720 or SD screen, still use the lower auto-resolultion.

That's my point - auto resolution *up to a given maximum*, which would be 1080p for low-power images.

natinusala, 2020-01-07:
4k is expected to be slow on a Raspberry Pi 4, there's really nothing we can do against that

Pxtl, 2020-01-07:
@natinusala - totally agree.  That's why imho it shouldn't be the default resolution when you plug in a Lakka Pi4 into a 4K TV.

RobLoach, 2020-01-08:
It's difficult to know what resolution devices will be using. I have a Raspberry Pi with Lakka plugged into a 4K TV, and another using a 480x320 display.

I like having the higher resolution, and it works for the core I'm using, since they're not hardware-intensive. This could be a core problem? Which PS1 core are you using?

By default, I would expect a PS1 core to use the native resolution of the PS1.

Pxtl, 2020-01-08:
Even emulating in native PS1 resolution, it's still rendering to the high-res viewport and getting performance problems.

closed 2020-08-18 by Ntemis

bagelbyheart, 2021-03-16:
Hey, I think this has more to do with performance at higher resolutions. For example, using an rpi3 at 1080p shows just a bit of FPS dip at the GUI while default settings, but runs fine at 960x540. The problem is that the default scaling used by the system in that case is bilinear.

The same issue occurs if you try to scale a display using xrandr with default settings, but can be worked around with a very simple fork of that. Of course since lakka doesn’t seem to use xorg or to have xrandr, I can’t really try that here, but it seems like the solution would be similarly applicable.

https://forums.linuxmint.com/viewtopic.php?t=159064