libretro/Lakka-LibreELEC #872: [Feature Request] 64-bit Kernel implementation

coldspark29, 2020-01-12:
How long would it take to implement the new 64-bit Kernel? 

https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=250730

I know it is still a beta version, but I guess it is early enough to start working on it. It will make N64 run faster and also provide support for Dreamcast's Windows CE games. It is also reported to keep the RPi4 much cooler!

Ntemis, 2020-01-12:
We are using a 64 bit kernel already with 32 bit userland driver.
Do you mean a full 64 bit image perhaps?

coldspark29, 2020-01-12:
Well, I mean whatever gets me Dreamcast's Windows CE games to run on my RPi4 :)

natinusala, 2020-01-12:
That is very unlikely to happen as the aarch64 dynarec is barely fast enough anyway

But yes the RPi image is entirely 32bit because they don't support 64bit userland drivers, there are ways to build and use them but they don't all work and they cannot all be built for 64bit

coldspark29, 2020-01-12:
@natinusala You mean the RPi4 is too slow? All other Dreamcast games run flawlessly. Even better than N64... 

So there is no chance, that Windows CE games will ever work?

natinusala, 2020-01-12:
Windows CE games are known to be harder to emulate

coldspark29, 2020-01-12:
I mean is it technically impossible or does it lack the processing power? I am only interested in Worms Armageddon really and that should be doable. Code Veronica is probably harder, I guess.

natinusala, 2020-01-12:
It's not impossible it might just not run as well as other Dreamcast games

coldspark29, 2020-01-12:
Well, Worms Armageddon hangs at the logo screen. Someone from Retroarch said it was due to the Pi4 running with a 32-bit OS

natinusala, 2020-01-12:
Well the aarch64 dynarec may be able to run the game better than the armv7 one but if the game freezes that sounds like a bug to me

coldspark29, 2020-01-12:
Okay, there was someone experiencing the same bug on his PC. They said they fixed it a few weeks ago. So I hope it will work with the next release of LAKKA. Any ideas when you will push it? Good to hear that there is hope!

natinusala, 2020-01-12:
This is unrelated to Lakka, you should see if the bug is fixed in the core first because if it's not, updating Lakka won't change anything :D

coldspark29, 2020-01-12:
So LAKKA does automatically update the cores or can I do it somehow? They said the Flycast update is already pushed

natinusala, 2020-01-12:
Oh then the next Lakka update will contain the updated core

coldspark29, 2020-01-12:
Thought so ^^

coldspark29, 2020-01-12:
So do you already have an idea when the next update is being pushed?

coldspark29, 2020-01-12:
Well, on the PC everything works fine. Maybe they are right and it is related to the lack of a 64-bit environment. I mean the Flycast team probably knows best. If no 64-bits are coming to the Pi, I guess I can say goodbye to Windows CE games. Most of them exist on PSX anyway and the rest of them are not necessarily the best titles. It is not so dramatic

natinusala, 2020-01-12:
We aim to release the update by the end of the week

coldspark29, 2020-01-12:
Okay, we'll see then, I guess. Thanks a lot for the info!

closed 2020-08-18 by Ntemis