libretro/Lakka-LibreELEC #873: "Enable SSH" Menu item does not start the ssh service on Pi Zero

Veritas83, 2020-01-16:
Flashed Lakka 2.3.1 to sd card. Booted, Filesystem expanded.

Configured controller. Entered WiFi Config. Flipped Enable SSH and Enable Samba buttons.

Rebooted.

Wifi does not automatically connect. Selecting WiFi it remembers and goes online.

Samba was accessible. SSH was not.

ToKe79, 2020-01-16:
can you elaborate 

> SSH was not

how did you try? command line / PuTTY / other? via hostname or IP? did you check if port 22 was accessible/open/listening (e.g. nmap, telnet from your computer)?

WiFi would eventually connect after the reboot, but it is faster to go to WiFi settings, it pokes `connmanctl`. Reboot was not necessary for SSH/Samba to start (at least from my experience / how the systemd units are set up).

Veritas83, 2020-01-18:
I used both nmap and PuTTY. I was only nmap'ing for -p 22, but i did notice the Raspberry MAC Address. (TCP/22 did not show up) It was responding to ping and I could see/browse the shares over smb.

I also tried creating an empty 'ssh' file in /boot/, that did not get sshd to start either

natinusala, 2020-01-19:
> I also tried creating an empty 'ssh' file in /boot/, that did not get sshd to start either

That is normal, Lakka doesn't use that at all

Can you use samba to see if the file `/storage/.cache/services/sshd.conf` exists? Hopefully `.cache` is exposed in the samba drive

ToKe79, 2020-01-19:
Unfortunately, we do not expose `/storage/.cache` via samba (source: https://github.com/libretro/Lakka-LibreELEC/blob/master/packages/network/samba/config/smb.conf)

However, you can check the content of the SD card on any extfs capable system - look for the `LAKKA_DISK` partition, folder `.cache/services`. If there is no `sshd.conf` file, just add an empty one. Then put the SD card back to your device and try booting again.

Or try to force the sshd start by adding `ssh` to `cmdline.txt` (at the end of the line, watch out for any auto-wrapping - it should be all just one line without any newlines).

Veritas83, 2020-01-19:
Currently in the middle of a long compile process on another (RetroPie buster) build on a diff sd card. Once that is done, I can reflash Lakka on a 16GB and I will try these suggestions.

closed 2020-08-18 by Ntemis