libretro/Lakka-LibreELEC #874: Build failure due to wrong package name for distro

mrVectorz, 2020-01-18:
- Which version of Lakka are you using?
master branch

- What system hardware are you using?
Not related to the issue, but a Lenovo T460p.
The issue is more distro related, which `Fedora 31`

- What did you do?
`DISTRO=Lakka PROJECT=RPi2 BOARD=RPi4 ARCH=arm make image`

- What did you expect to happen?
Give me the proper package requirements that I need to install.

- What happened instead?
~~~
**** Your system lacks the following tools needed to build Lakka ****
lsdiff gperf makeinfo glibc-static libstdc++-static /usr/include/ncurses.h perl/JSON.pm provided by patchutils gperf texinfo glibc-static libstdc++-static ncurses-devel libjson-perl
**** You seem to use a fedora system ****
would you like to install the needed tools ? (y/n) y
Last metadata expiration check: 1:25:23 ago on Sat Jan 18 11:46:45 2020.
No match for argument: libjson-perl
Error: Unable to find a match: libjson-perl
make: *** [Makefile:29: image] Error 1
~~~

To get around the issue I manually installed `perl-JSON` package and then I passed the required test:
`scripts/checkdeps`
~~~
if ! perl -MJSON -e 'exit(0);' 2>/dev/null; then
  files="$files perl/JSON.pm"
  files_pkg="$files_pkg libjson-perl"
fi
~~~

I would suggest making this also DISTRO related

Ntemis, 2020-01-18:
We don't support other distro than debian/ubuntu for building our images.
Reason: Linux distros is a rabbit hole and we lack the arms to get back to the ground

natinusala, 2020-01-19:
Yeah sorry you are on your own for that one. We cannot support each and every Linux distribution as Deme said it would be unmaintanable

closed 2020-01-19 by natinusala

mrVectorz, 2020-01-20:
I agree it would be difficult to maintain all of the. But doing this change is quite minor and simply adding that pkg to the build deps (per DISTRO which you already maintain) would just made it that much simpler.