libretro/Lakka-LibreELEC #875: PPSSPP core crashing  Generic PC Nightly

Gitlab1111, 2020-01-20:
CPU: Intel Haswell Intel© Core™ i5-4670 CPU @ 3.40GHz × 4
GPU: N/A
RAM: 4gb
SSD 60gb

I noticed for **Metal Slug XX** for PSP that it crashes sometimes upon mission completions and will crash if going to the Retroarch menu at times as well. Can someone clarify that PPSSPP **assets** help with better performance because before I never had that added to the Retroarch system PPSSPP folder before running any PSP games in the past.

Another Issue is that movie clips durring BlazBlue PSP games would get to the point of having a black screen and the Retroarch/lakka would then be rendered inoperative.

The past nightly versions also showed **missing assets** from games. I also had this issue with another game when using a **higher resolution**.

[Retroarch log](https://github.com/libretro/Lakka-LibreELEC/files/4083980/Retroarch.log)

Gitlab1111, 2020-01-20:
Not sure if its a Rom issue but there was also some graphical errors as well.
[image]
[image]

gouchi, 2020-01-20:
May you try to make a test with [this build](http://nightly.builds.lakka.tv/PRs/pr_838_000137/Generic.x86_64/).

Thank you.

Gitlab1111, 2020-01-22:
build /PRs/pr_838_000137/Generic.x86_64

Even with that build mention I still get the black screen which freezes retroarch/lakka and the movie clips like the one in the intro's still lag almost entirely before freezing and requiring to restart the who desktop pc.

Gitlab1111, 2020-01-24:
Still crashing when movie clips end on PSP even when using a different build. It crashed and didn't leave me any logs for it on the nightly Generic PC build of 1/24/2020.

emersonnovais, 2020-03-06:
I had this issues with "Metal Slug XX" and "Space Invaders Xtreme" in Ubuntu x64 since the 18.04 version.

gouchi, 2020-06-05:
May  you try to make a test with [this image.](https://nightly.builds.lakka.tv/members/gouchi/ra-1.8.7/Lakka-Generic.x86_64-2.9-devel-20200525183638-43653b3.img.gz)

Thank you.

gouchi, 2020-08-16:
Hi @emersonnovais,

Please try to make a test now with this [nightly build image.](https://nightly.builds.lakka.tv/2020-08-10/Generic.x86_64/Lakka-Generic.x86_64-3.0-devel-20200810061340-46502e1.img.gz)

Thank you.

emersonnovais, 2020-08-18:
Thanks, @gouchi. Since the last update before the annoying cracker sabotaged your repositories, they crashed any time. I will test now and give an feedback. Thank you so much. ; )

closed 2020-08-18 by Ntemis