libretro/Lakka-LibreELEC #876: Turbografx CD bin/cue games work?

Gitlab1111, 2020-01-20:
CPU: Intel Haswell Intel© Core™ i5-4670 CPU @ 3.40GHz × 4
GPU: N/A
RAM: 4gb
SSD 60gb

Nightly build 11/4/2009 Lakka OS

Does The core for Turbografx 16 cd's work with bin/cue games because the other ones in ccd seem to work fine except for bin/cue combo files. Logs 
[Retroarch TG16cd.txt](https://github.com/libretro/Lakka-LibreELEC/files/4084172/Retroarch.TG16cd.txt)

RobLoach, 2020-01-20:
Does the not-fast core one work?

Gitlab1111, 2020-01-24:
No since I only get the option to use The PCE Beetle Fast, it does not work. I have used other bios and switched between them with the same result in the end is a black screen. 
[NB Jan24 2020 t16cd.txt](https://github.com/libretro/Lakka-LibreELEC/files/4110938/NB.Jan24.2020.t16cd.txt)

I will try another bin/cue rom in a bit.

Gitlab1111, 2020-01-24:
The roms that were not working are working now with the new Jan 24,2020 nightly build for Generic PC. New Manual Scan works but it has the games double listed one in .cue and one in .bin. I wish I can screenshot it. 

The ones launched with .cue work now. Still cant get Exile2 rom to work. Not sure if that one is a bad rom or a special case.

[T16cd NB Lakka.txt](https://github.com/libretro/Lakka-LibreELEC/files/4111067/T16cd.NB.Lakka.txt)

closed 2020-08-18 by Ntemis