libretro/Lakka-LibreELEC #879: What are the standard settings for the gpu_memory section in config.txt?

coldspark29, 2020-01-22:
I know this is not really an issue in the right sense, but I did not know where else to ask for it.

Since I have a 4GB RPi4, I tried to increase the GPU memory, but the performance got worse. I just read that the GPU memory is only used for video decoding and the VPU uses its own memory handler, which uses the memory not used by the GPU. So increasing GPU memory means less memory for the VPU. Unfortunately I did not write down the standard settings and can also not find any info on it on the internet, so can you please tell me the values for

gpu_mem=
gpu_mem_256=
gpu_mem_256=
gpu_mem_1024=

ToKe79, 2020-01-22:
the default values can be found here, for example:
https://github.com/libretro/Lakka-LibreELEC/blob/bf8721be07755a8ab653a9b0941d6844651f9014/packages/tools/bcm2835-bootloader/files/3rdparty/bootloader/config.txt#L25-L31

More on `gpu_mem`: https://www.raspberrypi.org/documentation/configuration/config-txt/memory.md

closed 2020-01-22 by ToKe79

coldspark29, 2020-01-22:
Thanks a lot! I already searched, but it was dug too deep, I guess