libretro/Lakka-LibreELEC #880: Lakka bluetooth service fails after each reboot due to bluez.conf not found

matt404, 2020-01-23:
## Description

Hello, I'm encountering an issue that seems like a bug, and I'm surprised not to find anything about this on the google machine (and elsewhere), so maybe i'm doing something wrong?  Basically, my issue is that every time I turn on my lakka machine (running on a pi4), the bluetooth service won't start automatically.  I have configured bluetooth to start automatically with "systemctl enable bluetooth", but when I check the status of the bluetooth service I see the following failure reason:

 ConditionPathExists=/storage/.cache/services/bluez.conf was not met

When I create the bluez.conf file, everything works as expected, yay. But when I reboot the device, the bluez.conf file is deleted and the bluetooth service fails to start.  I then have to SSH into my lakka device to manually create the bluez file and restart the service.  It would be a nice upgrade to the experience and the security of the device if I didn't have to do this dance every time I turn on the device.

### Expected behavior

The bluetooth service should start automatically when lakka boots, assuming I set it to start automatically using "systemctl enable bluetooth"

### Actual behavior

The bluetooth service fails to boot.

### Steps to reproduce the bug

1. Turn on lakka device
2. Setup a controller via bluetooth, and add it as a trusted device
3. Enable the bluetooth service to start automatically
4. Reboot the device, and there you have it, blue tooth is busted when it should be working.

### Environment information

- Lakka Version: 2.3.1
- Hardware: Raspberry Pi 4

natinusala, 2020-01-23:
This is definitely a bug, we need to find out what deletes the file

Do you reboot properly?

matt404, 2020-01-25:
Hi, thanks for getting back so quickly.  It turns out I was being dumb and didn't notice that there is a setting in the Lakka UI to enable bluetooth... doh!  apparently if you have it disabled in the app, but try to enable it via command line, it will disable each time you reboot.  Sorry for the distraction, and thanks for your response.  It's working fine now that I've enabled it via the UI.

closed 2020-01-25 by matt404