libretro/Lakka-LibreELEC #882: PSP game loading doesn't work occasionally on 2.3.2 and RPi4

coldspark29, 2020-01-23:
I have had no issues with loading PSP games on 2.3.1. While performance has improved significantly on 2.3.2, sometimes loading games makes Retroarch crash and reboot. Might this be a Retroarch issue?

natinusala, 2020-01-23:
Do those games contain FMV? Do they crash when the FMV would normally play?

Gitlab1111, 2020-01-24:
Some logs would also help. To enable logs set it to create separate logs because for some odd reason nothing gets logged in retorarch.log. 2nd is to set both levels to **Debug 0**, I tried other settings in the logging menu but they don't work for some odd reason. Might need to enable show advanced options to get to the logging menu.

Your **logs** will be located here, I had to enable **root** for me to get to it. I usually copy and paste the contents from it to another empty text document because they usually end up being locked. (for both RPI4 and Generic PC)

[image]

coldspark29, 2020-02-01:
I enabled logs, but the folder is still empty. I have the same issue on an x64 PC with Gamecube games as well. The funny thing is, that it works for some cores if you have previously loaded an N64 game!!

On Raspberry it works for PSP and all the others

On Generic PC Gamecube works this way, but PSP does not work at all

coldspark29, 2020-02-09:
So I think I narrowed down the problem. I had to load an N64 game before loading PSP and Dreamcast games, because N64 turns off Run-Ahead. Dreamcast used to work with Run-Ahead on 2.3.1 I think. When I turn off Run-Ahead everything loads fine.

closed 2020-08-18 by Ntemis