libretro/Lakka-LibreELEC #883: Wrong screen scaling for N64 on 2.3.2 and RPi4

coldspark29, 2020-01-23:
For some games, e.g. Conker's Bad Fur Day, the scaling is wrong. For others, e.g. Mario Kart 64, it works fine. I've had zero issues on 2.3.1. I can set the overscan to match, but then it is wrong in the games, that work fine.

natinusala, 2020-01-23:
This is likely a new issue on mupen64plus_next, can you fill an issue on the mupen repo please? You'll probably need to give more details like what the exact core options you are using

coldspark29, 2020-01-23:
Will do. Thanks!

coldspark29, 2020-03-08:
I also have seen this issue now on Flycast. It occurs for MupenPlus and MupenPlusNext. I believe that this is a Lakka issue and would be really glad if it would be resolved before the release of Lakka 3.0

closed 2020-08-18 by Ntemis