libretro/Lakka-LibreELEC #884: Lakka 2.3.2. RPi1/0/GPi issues

brightentayle, 2020-01-23:
The new (supposedly) stable release of Lakka appears to be broken in a few places. I've tested GPi-2.3.2 specifically, and can confirm that lr-mgba crashes upon trying to start up.

There's also a glitch in User Interface -> Appearance -> Menu Color Theme (seems to happen in RGUI specifically) that doesn't have a name in the menu, appears as Anti-Zenburn on the item list and uses Classic Green's color scheme.

P. S. The new Accessibility feature can also not be hidden for whatever reason, even though it does nothing for Raspberry Pi-based builds, as they don't have any kind of TTS.

generalerusso, 2020-01-23:
Confirming the lr-gpsp crash-to-menu issue, all roms used (around 20) all seem to crash to the main menu after some 15-20 seconds wait.

generalerusso, 2020-01-23:
Another issue regarding lr-fbalpha2012 is when choosing UNIBIOS for the bios for playing NeoGeo roms. In Lakka v2.3.1 it will choose the most updated version in the neogeo.zip (generally 3.0), but in 2.3.2 it seems to load always Unibios 1.2.

natinusala, 2020-01-24:
Can you give me logs of the mgba and gpsp crashes please?

As for that BIOS issue I suppose that is a change in logic in RetroArch, can you reproduce it on desktop?

brightentayle, 2020-01-27:
gpSP in particular gives out this, which makes me think Lakka-RPi.GPICase.arm-2.3.2.img has this core compiled for the wrong architecture.

[image]

IndegoRune, 2020-04-30:
This is still an issue.

gouchi, 2020-08-16:
Hi,

Do you still reproduce your issue with this [nightly build images](https://nightly.builds.lakka.tv/2020-08-07) ?

Thank you.

IndegoRune, 2020-08-19:
How are you going to close this without any kind of resolution or answer? We have yet to see a new stable Lakka on GPi, and the old one is outdated as heck by this point.

You are asking if an issue is fixed in a nightly, when most of us likely don't mess with nightlies.

natinusala, 2020-08-19:
Nightlies are here to field test bug fixes and new features. I let you wait for the upcoming stable then.

Ntemis, 2020-08-19:
Also i need to add that  2.3.x is end of life so anyway you shouldnt expect any fixes to drop any time soon except the ones you will sent as a pr.

Ntemis, 2020-08-19:
@natinusala lesson learned we should stop merging hit and run pr's since we cant support the device we don't own.

brightentayle, 2020-08-19:
Okay, first of all, I have to say that I didn't get to test the nightlies yet, because I didn't have the time to do so. Second of all...

@Ntemis You do realise the GPi Case is not an emulation handheld of its' own, right? The GPi Lakka is basically the Raspberry Pi 0/1 build but slightly stripped down (yet still including lr-pcsx-rearmed, which doesn't even run at full speed even with frame skipping), with the safe shutdown patch and also the screen drivers included. So, aside from those three factors, all those issues me and a few others have described there would easily apply to the Pi 0/1 build of 2.3.2, because the thing **houses** a Raspberry Pi 0. If that makes any sense.

Ntemis, 2020-08-19:
I know what a GPi Case is but still changes nothing as i dont own any Pi 0/1 to test and make sure core is fixed.
If anyone else has it and can give it a shot surely be my guest and he/she can reopen this case.
For the time being please try to try the nightly images for the sake of all of us lets leave this release in peace and stop beating a dead horse. ofc feel free to fork and fix anything you like but before that again please try the latest nightly and if needed open a new issue so we can fix it this time

wfce, 2020-08-19:
If 2.3.2 is end-of-life and we should not be using it, please update your website saying as much.

Ntemis, 2020-08-19:
2.3.2 is tagged as a stable release that can be used until v3.x comes out, what you need us to say on the website? that you didnt test previous 2.3.x releases and you came on the last stable before transition to a new framework and demand a fix? where were you when releases were pouring hot one after the other? The train has departed so i suggest you move on too. Regards

wfce, 2020-08-19:
Nobody is demanding anything. 

We downloaded the version available on the website and found an issue, so we reported it.

Ntemis, 2020-08-19:
Thank you for your contribution but unfortunately there wont be any new versions released from this branch as it has being decommissioned. 
Your interest in transforming Lakka OS into a better distro is highly appreciated.
All efforts are now moved on for next major release and we continue passionate and focused to deliver the best RetroArch Turn-key Distro OS.
Feel welcome to test and report any issues by testing the new nightly images from here  https://nightly.builds.lakka.tv/

brightentayle, 2020-08-23:
Can confirm, it crashes with the same error on the most recent nightly.

...also, I'd really advise people to stop throwing corporate excuses around and at least buy the platforms they're developing for, because begging people to see if something works really isn't going to make debugging any easier. _Especially_ if those platforms cost five to ten bucks.

natinusala, 2020-08-23:
RPi0 support was brought by an external contributor who's no longer around. There is nobody to maintain it anymore The end result is that we're going to drop support for that board, period.

brightentayle, 2020-08-23:
@natinusala Thanks for clarifying that

natinusala, 2020-08-23:
It might be worth to continue support it if we get hold of a GPICase, but other than that I don't know if the rpi0 is still a relevant board anymore.

Ntemis, 2020-08-23:
Not only that but that core is failing everywhere as i read so is not a lakka issue at all.
Is correctly reported here https://github.com/libretro/gpsp/issues/71
This issue stays closed end of story.

Ntemis, 2020-08-24:
I went ahead and fixed it but you guys should wait for the new nightly to let me know

john-parton, 2020-08-24:
I have the GPi Case, and it's a really lovely piece of hardware. I understand that the Zero is basically ancient at this point, but it's still a versatile little device.

I'm a web developer, so this isn't my usual wheel house, but I can definitely test things out.

I'll try to test that fix tomorrow.

SethBarberee, 2020-08-24:
Compiling from master with recent commit and running on RPi0 just fine with gpSP. Latest nightly should be fine.

IndegoRune, 2020-08-24:
Cool!

john-parton, 2020-08-24:
I'm not seeing the latest nightly.

Don't see it here: http://nightly.builds.lakka.tv/2020-08-24/

The log says this: http://nightly.builds.lakka.tv/2020-08-24/_logs/nightly_2020-08-24.txt

```
Following targets were skipped - could not find existing build folder:
A64.arm
H3.arm
H6.arm
AMLGX.arm
Generic.x86_64
Generic.i386
MiQi.arm
RK3328.arm
RK3399.arm
TinkerBoard.arm
Gamegirl.arm
GPICase.arm
RPi.arm
RPi2.arm
RPi4.arm
```

Not familiar with the build system. Why would this happen?

SethBarberee, 2020-08-24:
@john-parton Probably due to the nightly build cleanout. Sometimes we have to cleanout the nightly build manually. I compiled my working image from commandline.

For RPi0/RPi1:
`DISTRO=Lakka ARCH=arm PROJECT=RPi DEVICE=RPi make image`

For GPiCase:
`DISTRO=Lakka ARCH=arm PROJECT=RPi DEVICE=GPiCase make image`

metchebe, 2020-08-24:
Hi @SethBarberee, just curious, does the fixed gpSP have the dynarec included?

SethBarberee, 2020-08-24:
Hmm looks like it wasn't included since platform was changed to only `unix`

Ntemis, 2020-08-24:
Issue is now reported fixed, so lets close it. Thanks for the reports

closed 2020-08-24 by Ntemis