libretro/Lakka-LibreELEC #885: Wifi fails randomly on RPi4 after upgrading to 2.3.2

jstolarek, 2020-01-23:
I just upgraded Lakka on my Raspberry Pi 4 to latest stable version 2.3.2. After the update WiFi stops working randomly. This means that it works on boot but then just drops and there seems to be no way to re-enable it.

What would be a civilized way of attaching a log (~750 lines). Not that it contains anything about Wifi.

natinusala, 2020-01-24:
This is strange because 2.3.2 didn't touch anything WI-Fi related. We would need to connman logs rather than the RetroArch logs to diagnose this

jstolarek, 2020-01-24:
How can I get connman logs? I can see them when viewing `journalctl` but obviously the moment wifi dies I can no longer see logs for the current session. I need a way to persist them between the boots. On a standard Linux distro I could modify the config to tell journald to put logs in a file but it seems that the system partition on Lakka is created on the fly on boot and there is no way to modify the config. My Linux knowledge is useless here.

natinusala, 2020-01-25:
You can use Ethernet to re-reconnect and grab the logs

jstolarek, 2020-01-25:
Are there any alternatives? Getting the Pi connected via Ethernet in my flat requires a major effort. I might find the time to do it one day but not any time soon.

natinusala, 2020-01-25:
I know that struggle 😅 

The other alternative is a dumb script and service that outputs the journald result to a file every 10s or so

jstolarek, 2020-01-26:
So the logging method works but now no matter how hard I try I can't reproduce the problem :-/

closed 2020-08-18 by Ntemis