libretro/Lakka-LibreELEC #889: Pc x64 Lakka 2.3.2 Dolphin Emulateur crash

andre726, 2020-01-24:
I was on 2.3.1 version and all my gamecube launch well, i do the recently update and now when i launch dolphin emulation it's crash and i have a black scren. I have to reboot the computer.

maybe i can watch the log to help you

natinusala, 2020-01-24:
Can you send us the logs please?

andre726, 2020-01-24:
okay, can u tell me where i can find it?

natinusala, 2020-01-25:
The Lakka wiki has everything you need for that :D

Gitlab1111, 2020-01-27:
You may not like the response but you can't expect a team of 5 to do everything for everyone or hold their hands through everything. Some of the issues are likely user error due to lack of knowledge and inexperience. 
 Logs are located in /Storage/.config/retroarch/logs  I use my **Linux distro** to open it up because it requires **root access** for me to get into it.

If you are not getting any logs recorded then try setting it to verbose logging and setting both levels to **debug 0**.

or follow Natinusala's advice for a more reliable way of getting logs recorded.

"The preferred (and only reliable way) of getting logs is to run RA manually from command line with the --verbose option and redirect the output to a file. This way the file won't be deleted if RA crashes (since the terminal writes it)."

@andre726 yes I noticed that too when I updated to newer build that my **gamecube** emulator has been **crashing** as well.

Gitlab1111, 2020-01-27:
@andre726 @natinusala  I do tried to run it on a rom that was working fine with a previous build. It ends up as a black screen and stays frozen from there.  logs

[Lakka Dolphin core.txt](https://github.com/libretro/Lakka-LibreELEC/files/4115669/Lakka.Dolphin.core.txt)

natinusala, 2020-01-27:
It looks like something broke OpenGL on the core side, can you try to reproduce on desktop, possibly with the same GPU driver?

andre726, 2020-01-28:
Hello this my log file, i'm running lakka on a macbook pro from 2012

[retroarch.log](https://github.com/libretro/Lakka-LibreELEC/files/4121330/retroarch.log)

Mr-Jake, 2020-01-28:
Posted on the Libretro forum by hunterk:

> Go to settings > core > allow cores to change the video driver OFF (if you don’t see settings > core submenu, go to settings > user interface > show advanced settings ON).

https://forums.libretro.com/t/dolphin-not-working-on-1-8-3-stable-win10/26079

natinusala, 2020-01-29:
Is Dolphin really changing the video driver? It should change from gl to gl anyway, I don't understand why it matters

Mr-Jake, 2020-01-29:
@natinusala

https://github.com/libretro/dolphin/issues/124

natinusala, 2020-01-29:
> libretro/dolphin#124 <https://github.com/libretro/dolphin/issues/124>

Oh so it's a known issue with the core, gotcha
Graceful driver switching should probably be disabled by default on Lakka
then

andre726, 2020-01-29:
great, it's work very well for me thanks

gouchi, 2020-08-16:
I will close this issue as workaround was provided.

closed 2020-08-16 by gouchi