libretro/Lakka-LibreELEC #890: Wifi names (SSID) truncated in the menu

jstolarek, 2020-01-25:
The wifi names (SSID) are truncated in the Lakka configuration menu. My Wifi router provides several networks at different frequency bands and I have to guess which one I am connecting to because the name is not displayed in full.

ToKe79, 2020-02-12:
How many letters are displayed?

RobLoach, 2020-02-12:
Could be an issue with the XMB menu? It should scroll the menu items. Screenshot?

jstolarek, 2020-02-13:
[image]

That's 19 or 20 characters - there's a space after the `1Gb` part in the SSID.

ToKe79, 2020-02-17:
I see - the ssid lenght is hardcoded - array of 20 chars:
https://github.com/libretro/RetroArch/blob/903648f1fe28c26e9c8c9dc59062a3853c6f4779/wifi/drivers/connmanctl.c#L198
I think we can change it to 32, although there is no official document (that I am aware of) which defines the maximum length of an SSID.

RobLoach, 2020-02-18:
https://routersecurity.org/SSID.php

> The maximum length of a WiFi network name is 32 bytes/characters.

Yup! Should be fixed on the next Lakka update. Good find :+1:

closed 2020-02-18 by RobLoach