libretro/Lakka-LibreELEC #891: Opening Capcom vs SNK 2 (flycast core) freezes RetroArch

jstolarek, 2020-01-26:
Attempting to load Capcom vs SNK 2 with flycast core freezes/crashes the RetroArch service. The service can be restarted using `systemctl`. Happens both with arcade and Dreamcast version of the game. System logs corresponding to the event say:

```
Jan 26 18:54:33 Lakka kernel: Alignment trap: not handling instruction f44c2add at [<abbd05b8>]
Jan 26 18:54:33 Lakka kernel: Unhandled fault: alignment exception (0xa21) at 0x01da8c54
Jan 26 18:54:33 Lakka kernel: pgd = b612c3c9
Jan 26 18:54:33 Lakka kernel: [01da8c54] *pgd=0ba72003, *pmd=fba23003
Jan 26 18:55:40 Lakka kernel: Alignment trap: not handling instruction f44c2add at [<abefc5b8>]
Jan 26 18:55:40 Lakka kernel: Unhandled fault: alignment exception (0xa21) at 0x010f4c54
Jan 26 18:55:40 Lakka kernel: pgd = 91ee722c
Jan 26 18:55:40 Lakka kernel: [010f4c54] *pgd=0125b003, *pmd=fbedf003
```

Platform: Lakka 2.3.2. + RPi4. Same roms work without problems on RA 1.7.3 + flycast on x86_64 platform (pure RA, not Lakka).

natinusala, 2020-01-27:
Try on desktop with the same flycast version to make sure that it's been fixed since

jstolarek, 2020-02-01:
By "desktop" do you mean getting a [generic Linux build](http://www.lakka.tv/get/linux/generic/), putting it on a USB, and booting off the USB?

natinusala, 2020-02-03:
I mean a desktop distribution of your choice like Ubuntu or Debian

jstolarek, 2020-02-03:
I have no problem running CvS2 on flycast core on a desktop (Debian 10, RetroArch 1.7.3).

jstolarek, 2020-02-03:
That is essentially the reason why I filed a bug report on Lakka and not flycast.

lakkanachronism, 2020-05-01:
[log.txt](https://github.com/libretro/Lakka-LibreELEC/files/4566421/log.txt)
I have Lakka 2.3.2 running on a Raspberry Pi 4 Model B. Every time I attempt to run Sonic Adventure 2, I get a black screen, and then the Lakka menu reappears. It seems that the flycast emulator doesn't even start because the Dreamcast startup screen never appears and the log says that the emulator runs for 0 seconds.

closed 2020-08-18 by Ntemis