libretro/Lakka-LibreELEC #896: ds emulation

Tatespi, 2020-02-04:
Hi guys 
been using retro arch on my msi dc windbox 100 for around 8 months now. it runs psp and n64 at 60fps but wont run ds games at 45 fps even. is there a way to make the fps better or reset the core settings as im sure I had ds emulation running at 60fps at one point
thanks

natinusala, 2020-02-05:
Hello,

DS emulation is popular for being slow on approximately all platforms. You can have it running with a beefy PC (which is not what you have), but don't expect steady 60FPS anytime soon

RobLoach, 2020-02-05:
I've had to decrease the resolution and switch up shading options to get it running smoothly enough.

Gitlab1111, 2020-02-07:
What core are you using? For some reason Melon DS which requires the NDS bios seems to run better than the other 2 which use HLE to run NDS games.

natinusala, 2020-02-07:
Honestly Desmume isn't expected to run fast on any platform but 32bit arm, melonDS is your best bet

Mr-Jake, 2020-02-07:
Exophase stated he will make DraStic open source soon. Hopefully this will allow a Libretro core to be ported.

natinusala, 2020-02-10:
Let's hope so

closed 2020-08-18 by Ntemis