libretro/Lakka-LibreELEC #899: [Flaycast Bug] Joystick movement triggers D-pad buttons as well.

blue-agent, 2020-02-15:
basically when playing "record of lodoss war" in flycast, with the latest 2.3 build of lakka for the RPI3B+, i discovered that with a generic USB DxInput gamepad, moving the joystick UP would cause the D-pad to go UP at the same time, and conversely, moving the joystick DOWN would also cause the D-pad to go DOWN as well.

the adverse affects of this is that when you make your character walk upwards in lodoss war, the cursor in the items bar on the right side of the screen, will scroll upwards indefinitely until you stop moving, making it impossible to use any items.

this bug also persists in Macross M3, when you specifically move the joystick UP or DOWN to fly the plane, this will trigger the UP and DOWN button on the D-pad as well, causing the plane to transform into robot mode uncontrollably when flying. (afterall the D-pad is responsible for plane transformation) thus making the game unplayable.

i've tried everything to fix the bug, including reconfiguring the controller or tried using different controller, but nothing works. i also tried one of the nightly builds, i don't remember which version as i have already deleted it due to the problem still being there.

for other games like "evolution the world of sacred device" where the D-pad are programmed to functions like the joystick....well due to the joystick and d-pad syncing together in flycast, it would also resulted in sudden and erratic movement.

i would be happy to do further testing, please let me know anytime.

natinusala, 2020-02-15:
Can you maybe try disabling the analog to dpad feature in the input settings?

blue-agent, 2020-02-16:
i disabled analog to dpad config a lonnng time ago, but it never worked.

as i mentioned earlier, i plugged the same usb controller into my pc and ran reicast, this bug is also present in the windows version of reicast. (build date: may 2019) 

i also tried redream in windows, again using the same usb controller, and the bug is not there. 

so i have ruled out the controller being at fault. the bug is only present in windows version of reicast and raspberry pi version of flycast.

did you encounter this bug on your pi? if not then could you tell me which version of lakka are you running on so i can download that same version of lakka that you have and do further testing?

blue-agent, 2020-02-16:
i've doubled checked the input settings, here's what it shows:

User 1 device type: Retropad
User 1 analog to digital type: none

is this the one you were referring to? i already setted this but it doesn't seem to fix anything.

blue-agent, 2020-02-16:
by the way i need to emphasized that i only brought up reicast and redream as proof that my controller is not at fault, as the issue of my controller was brought up and i answered to the best of my ability.

when troubleshooting i always give out many details as i can, unlike others who make low effort posts with vague questions full of typos.

 with that said, again, i only talked about reciast in the context that the bug is there also, to stressed that there might be a possibility that the bug was passed down from reicast to flycast. as for redream, i talked about that in the context of the bug not being there, even though i'm using the SAME usb controller which enountered problems in flycast of Lakka PI3. 

with that said, i hope i have clear things up for everyone. i want you all to know that there is absolutely no malice involved on my end. i'm not advertising reicast or redream or putting it on a high pedestal.  everything i say, i say for the sake of troubleshooing. again, let me stressed that there is no malice involved on my end.

blue-agent, 2020-04-19:
UPDATE:

i think i figured out what the problem is.

in lakka, the right joystick of my controller is seen as an duplicate D-PAD
also, the left joystick of my controller is seen as duplicate ABXY button set.

i do not know why lakka sees my controller like that, and configured it as such by default, but i am not at fault for this.

i tried the same controller in redream and macross m3 works flawlessly, with the right joystick being seen as a joystick and not a duplicate d-pad. thus i can fly the plane with the joystick and transform into robot mode using the d-pad.

here's the controller i have.

https://www.amazon.com/Controller-Gamepad-Joystick-Joypad-Computer-xBox360/dp/B00IXHP4IK
[image]

i managed to figure this out all on my own, by myself, because while i'm providing as much useful information as i can, i have received no communication from anyone here, nor did anyone requested me to supply more information.

 to save money, i've had to cut off my home internet contract since april 11 this year, then i bought a 5gb data 3G sim card in order to get online and post this information for you. please understand that i am not lying and that this is a real problem i'm trying to report.

closed 2020-08-18 by Ntemis

OctopusButtons, 2022-06-12:
This issue still happens, I’m in Flycast 22ad95c in RetroArch 1.10.2 on Mac (non-metal version).

if you bind d-pad left or right in RetroArch, then whatever buttons you binded for Analog Stick Left/Right will wrongly also activate d-pad left/right input.  The only way to avoid the problem is to unbind D-Pad left/right to nothing.

In Metropolis Street Racer the problem means that all steering inputs activate the HORN HONK (because the game assigns that to left/right dpad).

- I made sure that Analog to Digital Type = None.  Didn’t help
- I unbound and rebound everything, didn’t help
- I delete core and game remap files, didn’t help
- I tested in stand-alone Redream emulator and the controls are normal and correct
- I also tested in Street Fighter Alpha 3 in Flycast, but the issue is apparently moot because it doesn’t matter whether dpad and analog inputs are being input at the same time, there’s no adverse effect.  Unlike in MSR (horn honk) or OP’s game menu issue.
- I also tried binding Dpad Left/Right to real-world controlller UP and DOWN dpad to see if Analog left right and Dpad left right were still entangled.  They were.

In other words: **Analog stick left/right always secretly automatically activates DPAD LEFT/RIGHT, if Dpad left/right is bound to anything.** Only way to fix is to completely unbind Dpad Left/Right so that no button no I out (“- -“) is assigned.

Ntemis, 2022-06-12:
You better report it upstream then as its def not a lakka issue

OctopusButtons, 2022-06-12:
> 

Yes sorry about that, it really seems like the page suddenly changed to Lakka but previously said Flycast or RetroArch or a different area, strangely.  Looking for correct place now.