libretro/Lakka-LibreELEC #900: Raspberry Pi 4 directory save bug

Cibomatto2002, 2020-02-17:
When changing the directory to save to the USB the Saves and Savestates will forget when you do a reboot and you have to set it again when booting into Lakka. Using 2020-01-24 build.

ToKe79, 2020-02-17:
can you try following as workaround if this works:
Change the setting you want. Instead of "Shutdown" select "Restart RetroArch" from the "Lakka" menu. It should then quit and start again. Let me know if the setting was preserved.

Cibomatto2002, 2020-02-17:
It was not preserved. It shows Content dir. 
[image]

ekx, 2020-06-30:
Also happens to me on x86_64 on a Lenovo m93p tiny. Additionally it will not be set to default on reboot, instead it will be set to "<content directory>"

Cibomatto2002, 2020-07-20:
Just downloaded 2020-07-19/RPi4.arm nightly the bug is still there.

Ntemis, 2020-08-20:
@Cibomatto2002 can you try https://nightly.builds.lakka.tv/2020-08-20/RPi4.arm/Lakka-RPi4.arm-3.0-devel-20200820023653-59d5c4b.img.gz please

Cibomatto2002, 2020-08-21:
Still not saving to my usb 
[image]

Cibomatto2002, 2020-08-23:
Tried the Aug 23 build bug still there.

Cibomatto2002, 2020-09-23:
tried it with another usb stick and now it works :)

closed 2020-09-23 by Cibomatto2002