libretro/Lakka-LibreELEC #904: Lakka-RPi4.arm-2.9-devel-20200213010100-db102f6.img.gz not compatible with Raspberry Pi 4?

coldspark29, 2020-03-01:
I am getting the message that this image is not compatible with the hardware, when I load it in the update folder.

ToKe79, 2020-03-01:
That is true, this image is based on LibreELEC 9.2 (future Lakka 3.0, currently versioned as 2.9). In the now 'legacy' branch (Lakka-2.3.2), the RPi4 project has tag `RPi2.RPi4-arm` (as it is created as a "subdevice" of the RPi2 project), the current master branch tags the build as `RPi4-arm` and the compare check fails.
To work around this, put empty `.nocompat` file to the update folder (next to the `.img.gz` file) and it should work. However, as you are upgrading from Lakka 2.3.x, which is based on LibreELEC 8.2 branch, the results are not yet known. I am not sure if anybody tried that yet.

coldspark29, 2020-03-01:
I am updating from the former nightly build, which is identical to 2.3.2, I believe. For what is this build intended if not for upgrading from the former nightly?

coldspark29, 2020-03-02:
It worked like a charm.

ToKe79, 2020-03-02:
Glad it worked!

closed 2020-03-02 by ToKe79