libretro/Lakka-LibreELEC #907: Error building Citra

TiagoLuckner, 2020-03-06:
Trying to compile the image for Generic x86_64 and I'm stuck with the error:

[245/343] [DONE] install libpcap
BUILD      citra (target)
    TOOLCHAIN      cmake (auto-detect)
Executing (target): cmake -GNinja -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCMAKE_TOOLCHAIN_FILE=/home/tiago/Lakka-LibreELEC/build.Lakka-Generic.x86_64-2.9-devel/toolchain/etc/cmake-x86_64-libreelec-linux-gnu.conf -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=MinSizeRel -DENABLE_LIBRETRO=1 -DENABLE_SDL2=0 -DENABLE_QT=0 -DCMAKE_BUILD_TYPE="Release" -DBOOST_ROOT=/home/tiago/Lakka-LibreELEC/build.Lakka-Generic.x86_64-2.9-devel/boost-1_65_1 -DUSE_SYSTEM_CURL=1 -DTHREADS_PTHREAD_ARG=OFF -DCMAKE_NO_SYSTEM_FROM_IMPORTED=1 -DCMAKE_VERBOSE_MAKEFILE=1 --target citra_libretro /home/tiago/Lakka-LibreELEC/build.Lakka-Generic.x86_64-2.9-devel/citra-37192ff
CMake Error at CMakeLists.txt:46 (message):
  Git submodule externals/nihstro not found.Please run: git submodule update
  --init --recursive
Call Stack (most recent call first):
  CMakeLists.txt:51 (check_submodules_present)

CMake Error at CMakeLists.txt:46 (message):
  Git submodule externals/soundtouch not found.Please run: git submodule
  update --init --recursive
Call Stack (most recent call first):
  CMakeLists.txt:51 (check_submodules_present)

CMake Error at CMakeLists.txt:46 (message):
  Git submodule externals/catch not found.Please run: git submodule update
  --init --recursive
Call Stack (most recent call first):
  CMakeLists.txt:51 (check_submodules_present)

CMake Error at CMakeLists.txt:46 (message):
  Git submodule externals/dynarmic not found.Please run: git submodule update
  --init --recursive
Call Stack (most recent call first):
  CMakeLists.txt:51 (check_submodules_present)

CMake Error at CMakeLists.txt:46 (message):
  Git submodule externals/xbyak not found.Please run: git submodule update
  --init --recursive
Call Stack (most recent call first):
  CMakeLists.txt:51 (check_submodules_present)

CMake Error at CMakeLists.txt:46 (message):
  Git submodule externals/cryptopp/cryptopp not found.Please run: git
  submodule update --init --recursive
Call Stack (most recent call first):
  CMakeLists.txt:51 (check_submodules_present)

CMake Error at CMakeLists.txt:46 (message):
  Git submodule externals/fmt not found.Please run: git submodule update
  --init --recursive
Call Stack (most recent call first):
  CMakeLists.txt:51 (check_submodules_present)

CMake Error at CMakeLists.txt:46 (message):
  Git submodule externals/enet not found.Please run: git submodule update
  --init --recursive
Call Stack (most recent call first):
  CMakeLists.txt:51 (check_submodules_present)

CMake Error at CMakeLists.txt:46 (message):
  Git submodule externals/inih/inih not found.Please run: git submodule
  update --init --recursive
Call Stack (most recent call first):
  CMakeLists.txt:51 (check_submodules_present)

CMake Error at CMakeLists.txt:46 (message):
  Git submodule externals/libressl not found.Please run: git submodule update
  --init --recursive
Call Stack (most recent call first):
  CMakeLists.txt:51 (check_submodules_present)

CMake Error at CMakeLists.txt:46 (message):
  Git submodule externals/cubeb not found.Please run: git submodule update
  --init --recursive
Call Stack (most recent call first):
  CMakeLists.txt:51 (check_submodules_present)

CMake Error at CMakeLists.txt:46 (message):
  Git submodule externals/discord-rpc not found.Please run: git submodule
  update --init --recursive
Call Stack (most recent call first):
  CMakeLists.txt:51 (check_submodules_present)

CMake Error at CMakeLists.txt:46 (message):
  Git submodule externals/libzmq not found.Please run: git submodule update
  --init --recursive
Call Stack (most recent call first):
  CMakeLists.txt:51 (check_submodules_present)

CMake Error at CMakeLists.txt:46 (message):
  Git submodule externals/cppzmq not found.Please run: git submodule update
  --init --recursive
Call Stack (most recent call first):
  CMakeLists.txt:51 (check_submodules_present)

-- Target architecture: x86_64
-- Boost 1.66.0 or newer not found, falling back to externals
-- Target architecture: x86_64
-- Version: 6.0.0
-- Build type: "Release"
-- CXX_STANDARD: 17
-- Configuring incomplete, errors occurred!
See also "~/Lakka-LibreELEC/build.Lakka-Generic.x86_64-2.9-devel/citra-37192ff/.x86_64-libreelec-linux-gnu/CMakeFiles/CMakeOutput.log".
See also "~/Lakka-LibreELEC/build.Lakka-Generic.x86_64-2.9-devel/citra-37192ff/.x86_64-libreelec-linux-gnu/CMakeFiles/CMakeError.log".
FAILURE: scripts/install citra has failed!
[246/343] [FAIL] install citra

The following logs for this failure are available:
  stdout: ~/Lakka-LibreELEC/build.Lakka-Generic.x86_64-2.9-devel/.threads/logs/243/stdout
  stderr: ~/Lakka-LibreELEC/build.Lakka-Generic.x86_64-2.9-devel/.threads/logs/243/stderr

parallel: This job failed:
package_worker 3 243 343 'install citra'
Parallel build failure - see log for details. Time of failure: Fri Mar  6 16:18:45 -03 2020
make: *** [Makefile:9: release] Error 1

ToKe79, 2020-03-06:
Strange, `git submodule --init --recursive` should run during the `GET` stage. Do you have `git` installed?

ToKe79, 2020-03-06:
Sure you have, otherwise it would not clone.
Can you try `rm -rf sources/citra` and then build again?

TiagoLuckner, 2020-03-06:
I have git installed.

TiagoLuckner, 2020-03-06:
It works, tnks!

closed 2020-03-06 by TiagoLuckner