libretro/Lakka-LibreELEC #909: mGBA  jerky rewind sometimes.

Cibomatto2002, 2020-03-16:
RPi2.RPi4.arm-2.3.2.img

Pie 4

if I press rewind on some GBA games it will struggle to rewind this is the game I found the rewind problem on  Final Fight One (USA). There are other GBA game that do it to. just start playing the game then press rewind you will see it having trouble rewinding it's like it almost can't do it.

RobLoach, 2020-03-17:
Rewind is pretty CPU intensive, since it saves and loads every frame. I've seen this happen on my raspberry pis too. Surprised you got it to work at all :wink:

Cibomatto2002, 2020-03-17:
Do I need to close this then? I noticed the Mesen (NES Emulator) can barley run on Pie 4 and the rewind works correct even on the snes9x 1.60 it works. That's why I thought it was a bug in mGBA because those emulators it would work.

closed 2020-08-18 by Ntemis