libretro/Lakka-LibreELEC #912: ESpeak does not currently work in Lakka

liamerven, 2020-03-20:
As of Retroarch 1.82, a narration feature is now supported. This feature allows a visually impaired gamer to read RA menus along with using an AI translation serveice to read text in games. This requires Espeak which the current distros of Lakka do not ship with.

LibreElec contains an existing package for Espeak, but this is the old espeak hosted on Sourceforge. A newer package (Espeak NG) exists on Github which is regularly updated.

with retroarch stopped, Espeak Classic does not produce any audio. aplay, speaker_test, and even the -w (waveout) option of espeak all give us sound whent he wave is played through aplay.

This leads me to believe that there is something in the current espeak binary causing issues. there is no verbose logging for Espeak and Espeak itself does not give any errors.

I think this is important to look in to since the narrator feature is now a standard feature which ships in RA. As it stands, this is a feature of the software a user is unable to use. I'm not sure if this is a holy justifyable reason, but I feel it still should be looked in to and resolved. I understand this effects an extremely minor subset of users, but Espeak I imagine is weighted very low, and should not have any noticeable impact on performance when included in the main distro.

stas-prze, 2020-04-19:
thanks for letting me know about that.
I am now planning to get something accessible and portable i can play my ds/gba/other games on, and i have a spare pi i bought, but well. If lakka doesn't yet work with RA's accessibility features, i guess i have to play on my pc or find an functional distro for my pi that wil allow me to start RA with accessibility features on and ai service.
Really want to glitch some pokemon yellow, play through pokemon black2 and a few other games without just mashing A when text is displayed, but read it with OCR, and RA is the only emulator whitch has that stuff build in to it.

gouchi, 2020-05-18:
@liamerven may you try this [image](https://nightly.builds.lakka.tv/members/gouchi/912/Lakka-Generic.x86_64-2.9-devel-20200518143704-43653b3.img.gz) ?

Requirements
- audio_device = "" in `/storage/.config/retroarch/retroarch.cfg`
- Update/create `/storage/.config/asound.conf` with the alsa card id from `cat /proc/asound/cards` with
```
defaults.pcm.card X
defaults.ctl.card X
```

closed 2020-08-19 by Ntemis