libretro/Lakka-LibreELEC #913: n64 emulators don't work

Renetrox, 2020-03-20:
This issue tracker is NOT for help or support. Please use the Lakka forum (https://forums.libretro.com/c/libretro/lakka-tv-general) or IRC @ #lakka on freenode.net for those kinds of posts.

BEFORE you report a bug make sure you have tried the latest nightly version of Lakka at http://le.builds.lakka.tv/. Your bug might be already fixed.

If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
2.3.2
- What system hardware are you using?
raspberry pi 3b
- What did you do?

the n64 emulator does not load any games (Mupen64 plus Next, Parallel n64) and the videos (ffmpge) do not play, they cause the system to freeze
- What did you expect to happen?
The emulator I used was mupen64plus-Next and Parallel N64, it doesn’t open any games. Video: ogg, mpge, mp4, etc.
- What happened instead?
the system  freeze

Mupen64plus-Next

[INFO] RetroArch 1.8.4 (Git 9e25601a56)
[INFO] === Build =======================================
[INFO] Capacidades: NEON VFPv3 VFPv4
[INFO] Built: Jan 20 2020
[INFO] Version: 1.8.4
[INFO] Git: 9e25601a56
[INFO] =================================================
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[INFO] Versión de la API Libretro: 1
[INFO] Compilado para la API: 1
[INFO] [Audio]: Set audio input rate to: 30000.00 Hz.
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver …
[INFO] [EGL] Falling back to eglGetDisplay
[INFO] [EGL]: EGL version: 1.4
[INFO] [EGL]: Current context: 0x1.
[INFO] [GL]: Found GL context: videocore
[INFO] [GL]: Detecting screen resolution 1366x768.
[INFO] [EGL]: eglSwapInterval(1)
[INFO] [GL]: Vendor: Broadcom, Renderer: VideoCore IV HW.
[INFO] [GL]: Version: OpenGL ES 2.0.
[INFO] Querying EGL extension: KHR_image => exists
[INFO] [GL]: Using resolution 1366x768
[INFO] [GL]: Default shader backend found: glsl.
[INFO] [Shader driver]: Using GLSL shader backend.
[WARN] [GL]: Stock GLSL shaders will be used.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] Setting up menu pipeline shaders for XMB …
[INFO] [GLSL]: Compiling ribbon shader…
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Compiling simple ribbon shader…
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[... 253 lines cut ...]
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[WARN] [XMB] Asset missing: /tmp/assets/xmb/retroactive/png/disc.png
[INFO] [LED]: LED driver = ‘null’ 0x3a89c8
[INFO] [MIDI]: Initializing …
ALSA lib /home/natinusala/release/Lakka-LibreELEC/build.Lakka-RPi2.arm/alsa-lib-1.1.9/src/seq/seq_hw.c:466:(snd_seq_hw_open) open /dev/snd/seq failed: No such file or directory
[ERROR] [MIDI]: snd_seq_open failed with error -2.
[ERROR] [MIDI]: Initialization failed (list of input devices unavailable).
[INFO] Cargando historial: [/storage/.config/retroarch/content_history.lpl].
[INFO] Cargando historial: [/storage/.config/retroarch/content_music_history.lpl].
[INFO] Cargando historial: [/storage/.config/retroarch/content_video_history.lpl].
[INFO] Cargando historial: [/storage/.config/retroarch/content_image_history.lpl].
[INFO] Cargando favoritos: [/storage/.config/retroarch/content_favorites.lpl].
[INFO] [GL]: VSync => on
[INFO] [EGL]: eglSwapInterval(1)
[libretro INFO] mupen64plus: Game controller 0 (Standard controller) has a Memory pak plugged in
[libretro INFO] mupen64plus: Game controller 1 (Standard controller) has nothing plugged in
[libretro INFO] mupen64plus: Game controller 2 (Standard controller) has nothing plugged in
[libretro INFO] mupen64plus: Game controller 3 (Standard controller) has nothing plugged in
[libretro INFO] mupen64plus: Using CIC type X102
[INFO] [Environ]: SYSTEM_DIRECTORY: “/storage/roms/n64/”.
[INFO] [Environ]: SYSTEM_DIRECTORY: “/storage/roms/n64/”.
[INFO] [Environ]: SYSTEM_DIRECTORY: “/storage/roms/n64/”.
[INFO] [Environ]: SYSTEM_DIRECTORY: “/storage/roms/n64/”.
Segmentation fault (core dumped)

Renetrox, 2020-03-20:
Parallel

[INFO] RetroArch 1.8.4 (Git 9e25601a56)
[INFO] === Build =======================================
[INFO] Capacidades: NEON VFPv3 VFPv4
[INFO] Built: Jan 20 2020
[INFO] Version: 1.8.4
[INFO] Git: 9e25601a56
[INFO] =================================================
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[INFO] Versión de la API Libretro: 1
[INFO] Compilado para la API: 1
[INFO] [Audio]: Set audio input rate to: 30000.00 Hz.
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver …
[INFO] [EGL] Falling back to eglGetDisplay
[INFO] [EGL]: EGL version: 1.4
[INFO] [EGL]: Current context: 0x1.
[INFO] [GL]: Found GL context: videocore
[INFO] [GL]: Detecting screen resolution 1366x768.
[INFO] [EGL]: eglSwapInterval(1)
[INFO] [GL]: Vendor: Broadcom, Renderer: VideoCore IV HW.
[INFO] [GL]: Version: OpenGL ES 2.0.
[INFO] Querying EGL extension: KHR_image => exists
[INFO] [GL]: Using resolution 1366x768
[INFO] [GL]: Default shader backend found: glsl.
[INFO] [Shader driver]: Using GLSL shader backend.
[WARN] [GL]: Stock GLSL shaders will be used.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] Setting up menu pipeline shaders for XMB …
[INFO] [GLSL]: Compiling ribbon shader…
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Compiling simple ribbon shader…
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Compiling snow shader…
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Compiling modern snow shader…
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Compiling bokeh shader…
[INFO] [GLSL]: Found GLSL vertex shader.
[... 235 lines cut ...]
[INFO] [Font]: Using font rendering backend: freetype.
[WARN] [XMB] Asset missing: /tmp/assets/xmb/retroactive/png/disc.png
[INFO] [LED]: LED driver = ‘null’ 0x3a89c8
[INFO] [MIDI]: Initializing …
ALSA lib /home/natinusala/release/Lakka-LibreELEC/build.Lakka-RPi2.arm/alsa-lib-1.1.9/src/seq/seq_hw.c:466:(snd_seq_hw_open) open /dev/snd/seq failed: No such file or directory
[ERROR] [MIDI]: snd_seq_open failed with error -2.
[ERROR] [MIDI]: Initialization failed (list of input devices unavailable).
[INFO] Found #pragma parameter AA Input Res X (AA_RESOLUTION_X) 0.000000 0.000000 1920.000000 1.000000 in pass 0
[INFO] Found #pragma parameter AA Input Res Y (AA_RESOLUTION_Y) 0.000000 0.000000 1920.000000 1.000000 in pass 0
[INFO] Cargando historial: [/storage/.config/retroarch/content_history.lpl].
[INFO] Cargando historial: [/storage/.config/retroarch/content_music_history.lpl].
[INFO] Cargando historial: [/storage/.config/retroarch/content_video_history.lpl].
[INFO] Cargando historial: [/storage/.config/retroarch/content_image_history.lpl].
[INFO] Cargando favoritos: [/storage/.config/retroarch/content_favorites.lpl].
[INFO] [GL]: VSync => on
[INFO] [EGL]: eglSwapInterval(1)
[INFO] Written to playlist file: /storage/.config/retroarch/content_history.lpl
[libretro INFO] mupen64plus: Starting R4300 emulator: Dynamic Recompiler
[libretro INFO] mupen64plus: Init new dynarec
[libretro INFO] mupen64plus: ARM CPU Features:, Half, Thumb, FastMult, VFP, ESDP, NEON, VFPv3, TLS, VFPv4, IDIVa, IDIVt
[INFO] Written to playlist file: /storage/playlists/Nintendo - Nintendo 64.lpl
[libretro INFO] Glide64 ucode = 0xbc03e969
[libretro INFO] Microcode 0 - RSP SW 2.0X (Super Mario 64)
[libretro INFO] microcheck: old ucode: 2, new ucode: 0
Segmentation fault (core dumped)

RobLoach, 2020-03-22:
Rpi3 is pretty low powered for n64. I've had better experience on the 4.

Renetrox, 2020-03-23:
> Rpi3 is pretty low powered for n64. I've had better experience on the 4.

but, both emulators work in retropie for example ... in my case they stopped working when updating retroarch to version 1.8.4.

closed 2020-08-18 by Ntemis