libretro/Lakka-LibreELEC #918: PPSSPP video scenes playback lagging

iokzero, 2020-03-25:
All version of Lakka scines 2.3 + (including Lakka 3.0 Nightly)
Running on Intel i7 NUC7i7BNH NUC - CPU i7 7567U W/ intel iris plus 650 iGPU

Running PPSSPP, for games like God of War or Megaman, but the intro videos are very Laggy. The video will play a little fine, then lag for a sec, than play fine again for a sec or two. and repeat 

All version of Lakka with PPSSPP play fine with intro videos in games. 

I post the bug here sense the issues report for PPSSPP under Libretro is not longer there.

gouchi, 2020-08-16:
Hi,

Can you try this [nightly build image](https://nightly.builds.lakka.tv/2020-08-10/Generic.x86_64/Lakka-Generic.x86_64-3.0-devel-20200810061340-46502e1.img.gz) as we updated PPSSPP core and added some [patch to fix stuttering](https://github.com/libretro/Lakka-LibreELEC/blob/master/packages/libretro/ppsspp/patches/ppsspp-fix-fmv-stutter.patch) ?

Thank you.

iokzero, 2020-08-17:
Tried, It works! The stuttering is gone now.

FYI the emulating do some purses every once in a while at a random point for some reason (Can be in game or during video play back). But the video play back works. 

For whatever reason, this Nightly Build image also fix This issues #919

gouchi, 2020-08-18:
Thanks to @Ntemis who updated [PPSSPP core](https://github.com/libretro/Lakka-LibreELEC/commit/483bfe8a59b29f7f326d1e3eef71cd1edd884170) and provided a [patch](https://github.com/libretro/Lakka-LibreELEC/commit/d7117f7e7c3486518da8103cbe4d21ea927f9b56) ;-)

I will close this issue as it is fixed.

closed 2020-08-18 by gouchi