libretro/Lakka-LibreELEC #919: Some games wouldn't run after company logos on PPSSPP

iokzero, 2020-03-25:
All version of Lakka scines 2.3 + (including Lakka 3.0 Nightly)
Running on Intel i7 NUC7i7BNH NUC - CPU i7 7567U W/ intel iris plus 650 iGPU

Running PPSSPP, for games like Castlevania, the game just stop after the intro company logo shows up.

All version of Lakka with PPSSPP before Lakka 2.3 is fine

I post the bug here sense the issues report for PPSSPP under Libretro is not longer there.

Ntemis, 2020-08-11:
try this image
https://nightly.builds.lakka.tv/2020-08-10/Generic.x86_64/

closed 2020-08-11 by Ntemis

iokzero, 2020-08-12:
Would love to do some testing. Thanks

iokzero, 2020-08-12:
Tried the image you suggested can't get any game to launch With or Without the PPSSPP assists file in the system folder.

Ntemis, 2020-08-12:
Then is either a faulty iso or not supported by the emulator

iokzero, 2020-08-12:
They work just fine on LAKKA 2.2.2 ,is it possible that, the latest PPSSPP
no longer support .CSO format? Assuming no additional system files need to
be added from the user.

Ntemis, 2020-08-12:
please try the same game in iso format and let me know

iokzero, 2020-08-13:
I tried the original iso, PPSSPP failed to load as well, nothing launched.
I can't get a log out of it because I keep having ssh problems trying to
connect the box (For some reason always happens on nightbuild, not sure
why, the service is on for sure, but it keeps giving me "Host Key
verification failed..." error.)

iokzero, 2020-08-13:
Sorry I think I speak too soon, I finally got SSH working, forget to remove Keygen (LOL such 101 mistake, my bad)
`[INFO] [GL]: Found GL context: kms
[INFO] [GL]: Detecting screen resolution 3840x2160.
[INFO] [EGL] Found EGL_EXT_platform_base, trying eglGetPlatformDisplayEXT
[INFO] [EGL]: EGL version: 1.4
[INFO] [EGL]: Current context: 0x2611340.
[INFO] [KMS]: New FB: 3840x2160 (stride: 15360).
[INFO] [GL]: Vendor: Intel Open Source Technology Center, Renderer: Mesa DRI Intel(R) Iris Plus Graphics 650 (Kaby Lake GT3e) .
[INFO] [GL]: Version: 3.0 Mesa 19.3.5.
[INFO] [GL]: Using resolution 3840x2160
[INFO] [GL]: Default shader backend found: glsl.
[INFO] [Shader driver]: Using GLSL shader backend.
[INFO] [GLSL]: Checking GLSL shader support ...
[WARN] [GL]: Stock GLSL shaders will be used.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GL]: Using 4 textures.
[INFO] [GL]: Loaded 1 program(s).
[INFO] [GL]: Using GL_RGB565 for texture uploads.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Video]: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
[WARN] [udev]: Couldn't open any keyboard, mouse or touchpad. Are permissions set correctly for /dev/input/event*?
[INFO] [udev]: Pad #0 (/dev/input/event4) supports force feedback.
[INFO] [udev]: Pad #0 (/dev/input/event4) supports 16 force feedback effects.
[INFO] [Joypad]: Found joypad driver: "udev".
[INFO] [Video]: Found display server: null
[INFO] [Shaders]: Found shader "/tmp/shaders/bilinear.glslp"
[INFO] [Shaders]: Found shader "/tmp/shaders/nearest.glslp"
[INFO] [Shaders]: Found shader "/tmp/shaders/slang-conversion-template.glsl"
[INFO] [Shaders]: Found shader "/tmp/shaders/stock.glsl"
[INFO] ALSA: Using signed 16-bit format.
[INFO] ALSA: Period size: 1024 frames
[INFO] ALSA: Buffer size: 2048 frames
[INFO] [Display]: Found display driver: "gl".
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Display]: Found display driver: "gl".
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [LED]: LED driver = 'null' 0x900e40
ALSA lib /mnt/data/PROJECTS/Lakka-LE9.2/build.Lakka-Generic.x86_64/alsa-lib-1.1.9/src/seq/seq_hw.c:466:(snd_seq_hw_open) open /dev/snd/seq failed: No such file or directory
[ERROR] [MIDI]: snd_seq_open failed with error -2.
[ERROR] [MIDI]: Initialization failed (list of input devices unavailable).
[INFO] [SRAM]: SRAM will not be saved.
[INFO] [Playlist]: Loading history file: [/storage/.config/retroarch/content_history.lpl].
[INFO] [Playlist]: Loading history file: [/storage/.config/retroarch/content_music_history.lpl].
[INFO] [Playlist]: Loading history file: [/storage/.config/retroarch/content_video_history.lpl].
[INFO] [Playlist]: Loading history file: [/storage/.config/retroarch/content_image_history.lpl].
[INFO] [Playlist]: Loading favorites file: [/storage/.config/retroarch/content_favorites.lpl].
[INFO] [GL]: VSync => on
[INFO] [GL]: VSync => on
[INFO] [KMS]: New FB: 3840x2160 (stride: 15360).
[INFO] [KMS]: New FB: 3840x2160 (stride: 15360).
[INFO] [CONTENT LOAD]: Updating firmware status for: /tmp/cores/ppsspp_libretro.so on /tmp/system
[INFO] [CORE]: Using content: /storage/roms/Samsung USB/1229 - Castlevania - The Dracula X Chronicles (USA) (v1.02).iso.
[INFO] [CORE]: Arg #0: retroarch
[INFO] [CORE]: Arg #1: /storage/roms/Samsung USB/1229 - Castlevania - The Dracula X Chronicles (USA) (v1.02).iso
[INFO] [CORE]: Arg #2: -s
[... 97 lines cut ...]
[INFO] [DRM]: Mode 17: (1920x1080) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 18: (1920x1080i) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 19: (1920x1080i) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 20: (1920x1080) 1920 x 1080, 50 Hz
[INFO] [DRM]: Mode 21: (1920x1080i) 1920 x 1080, 50 Hz
[INFO] [DRM]: Mode 22: (1920x1080) 1920 x 1080, 30 Hz
[INFO] [DRM]: Mode 23: (1920x1080) 1920 x 1080, 30 Hz
[INFO] [DRM]: Mode 24: (1920x1080) 1920 x 1080, 25 Hz
[INFO] [DRM]: Mode 25: (1920x1080) 1920 x 1080, 24 Hz
[INFO] [DRM]: Mode 26: (1920x1080) 1920 x 1080, 24 Hz
[INFO] [DRM]: Mode 27: (1280x1024) 1280 x 1024, 60 Hz
[INFO] [DRM]: Mode 28: (1280x960) 1280 x 960, 60 Hz
[INFO] [DRM]: Mode 29: (1280x720) 1280 x 720, 60 Hz
[INFO] [DRM]: Mode 30: (1280x720) 1280 x 720, 60 Hz
[INFO] [DRM]: Mode 31: (1280x720) 1280 x 720, 50 Hz
[INFO] [DRM]: Mode 32: (1024x768) 1024 x 768, 60 Hz
[INFO] [DRM]: Mode 33: (800x600) 800 x 600, 60 Hz
[INFO] [DRM]: Mode 34: (720x576) 720 x 576, 50 Hz
[INFO] [DRM]: Mode 35: (720x480) 720 x 480, 60 Hz
[INFO] [DRM]: Mode 36: (720x480) 720 x 480, 60 Hz
[INFO] [DRM]: Mode 37: (640x480) 640 x 480, 60 Hz
[INFO] [DRM]: Mode 38: (640x480) 640 x 480, 60 Hz
corrupted double-linked list
Aborted
`

iokzero, 2020-08-17:
Nightly Build from #918 

Seems fix this issue as well for some reason.