libretro/Lakka-LibreELEC #921: Connecting to WPA2 Enterprise Networks

entr0p1, 2020-03-25:
Hi guys,

First of all thank you for an absolutely amazing project, it's by far the most polished of its kind I've ever seen and really elegant.

I'm trying to connect to a WPA2 Enterprise Wi-Fi network with my Lakka instance and when I use the menu, it's only prompting for a password as opposed to a Username and Password.

I understand this may not be a common request and probably isn't worth the dev time to implement at this stage, but if you could swing me in the direction of where I might place the wpa_supplicant.conf file so that Lakka picks it up on boot and connects, I'm more than happy to configure it myself.

- Which version of Lakka are you using?
   2.3.2
- What system hardware are you using?
   x86
- What did you do?
   Trying to connect to a WPA2 Enterprise network from the menu
- What did you expect to happen?
   Prompted for a Username and Password (WPA2 enterprise is user-based rather than a pre-shared key)
- What happened instead?
  Prompted for only a password

natinusala, 2020-03-25:
That would be a feature request for RetroArch, can you make an issue there ?

entr0p1, 2020-03-25:
Sure can, thank you!

closed 2020-03-25 by entr0p1

entr0p1, 2020-03-26:
For anyone who stumbles across this in future, to connect to an 802.1x WPA/WPA2 Enterprise network, do the following:

1. Store the CA certificate somewhere for the network (I used /storage/certs/<mycert>.cer)
2. Create a file at /storage/.cache/connman/<mySSID>.config and add the below contents:
```
[service_wifi_jupiter]
Type = wifi
IPv4 = dhcp
IPv6 = auto
Name = <SSID>
Passphrase = <PASSPHRASE>
Security = ieee8021x
EAP = peap
Phase2 = MSCHAPV2
Identity = <USER>
CACertFile = /storage/certs/<CA CERTIFICATE>.cer
```
3. Restart connman with `systemctl restart connman` and you should be connected

llozi, 2023-11-19:
This doesn't work for me on LibreELEC 11.0.3.
The only way I can connect to an 802.1x WPA/WPA2 Enterprise network (eduroam) is by putting an appropriate configuration file for iwd to /var/lib/iwd/eduroam.8021x additionally to the config file for connman described by [entr0p1](https://github.com/entr0p1) and then invoke `connmanctl connect wifi_ec086b19f395_....`.

The contents of /var/lib/iwd/eduroam.8021x looks like

```
[Security]
EAP-Method=PEAP
EAP-Identity=<anonymous@myuniversity>
EAP-PEAP-CACert=/storage/.config/cat_installer/ca.pem
EAP-PEAP-Phase2-Method=MSCHAPV2
EAP-PEAP-Phase2-Identity=<myidentity>
EAP-PEAP-Phase2-Password=<thepassword>

[Settings]
AutoConnect=true

```

No automatic connect is happening though, even when restarting connman service.
It looks like connman is not properly transmitting the configuration to iwd but iwd reads the configuration from the file instead when it is told to connect via connmanctl.