libretro/Lakka-LibreELEC #923: Wired controllers not working

bxyanks52, 2020-03-31:
This issue tracker is NOT for help or support. Please use the Lakka forum (https://forums.libretro.com/c/libretro/lakka-tv-general) or IRC @ #lakka on freenode.net for those kinds of posts.

BEFORE you report a bug make sure you have tried the latest nightly version of Lakka at http://le.builds.lakka.tv/. Your bug might be already fixed.

If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
- What system hardware are you using?
- What did you do?
- What did you expect to happen?
- What happened instead?

It is important to provide logs for debugging as part of your report. Instructions for generating logs can be found in the Troubleshooting Lakka documentation at http://www.lakka.tv/doc/Troubleshooting-Lakka/

Please delete this text before posting.

emilravn, 2020-04-10:
Currently also experiencing this. This happens after I start a game, it crashes (because of wrong core I suppose) restarts and unable to navigate the interface.
 
Tried deleting the retroarch.cfg through SSH and restore from new. Doesn't help.

Only thing that helps me is reinstalling Lakka itself.

Ntemis, 2020-08-11:
please upgrade to https://nightly.builds.lakka.tv/2020-08-07/RPi4.arm/

closed 2020-08-11 by Ntemis