libretro/Lakka-LibreELEC #924: System drive will unmounted if idle for too long

iokzero, 2020-03-31:
Lakka 2.2.2
Running on Intel i7 NUC7i7BNH NUC - CPU i7 7567U W/ intel iris plus 650 iGPU

Not sure if this still happening on new version, I haven't try it out yet because you need to leave Lakka for a long time without touching it, like 3~4hrs. And I don't have the most updated system install on my system drive because PPSSPP isn't right correctly on newer version. These are also several (maybe related) issues causing by idling the system for a long time on releases prior to Lakka 2.2.2, issues such as the ribbon will go crazy on XMB.

If you leave Lakka on for too long without touching it, all hard drive related function wouldn't work, such as: save state, load state, and playlists will start disappearing. If a game is loaded, it will still work for the time being, but as soon as you exit it, all your playlists are gone, and if you try to shutdown the system at that point, it will reload itself into XMB with no icon, no font, nothing except basic interface. And you can't shutdown the system from there either. 

If you do a hard restart. Everything will back to normal. 

Can someone confirm this bug on newer version of Lakka if you already got it installed on the system drive? 

This is a very weird issue, I have tried my best to describe it, but let me know if there are things I can clarify.

Many Thanks.

Ntemis, 2020-08-11:
2.x is eol

closed 2020-08-11 by Ntemis

iokzero, 2020-08-12:
Can I take it ask Lakka 3.0 is coming out? (Excited simile)