libretro/Lakka-LibreELEC #934: Wifi Error & CPU Information is unknown - x96 Mini [ROM16/RAM2] - Lakka 2.3.1 ARM 905

imadeput0287, 2020-04-16:
Hi,

I have Android Box X96 Mini 16GB ROM 2GB RAM version.
I was able to Install CoreElec 9.2.2 perfectly, but when I try to install Lakka 2.3.1 using dtb file :
- gxl_p212_2g.dtb (this one is from Lakka recommendation)
- gxl_p281_2g.dtb (this one is recommended by coreElec)
The wifi only show 1 SSID with weird name and not able to connect.

Not yet test for retro gaming.

Please help to fix, and please tell me what kind information needed to fix this.

Thank you.

Ntemis, 2020-04-16:
At the time of s905 porting to Lakka newer socs like yours didn't exist. So fundamentally the kernel doesn't cooperate so nicely with your system

imadeput0287, 2020-04-16:
I see, that sad to know the reality that lakka not yet support for S905W SOC.

Hope development team can add it up on next built. If any dump file or log needed maybe I can help.

And thank you.

Ntemis, 2020-04-16:
Best move is to switch to coreelec kernel and add support for all S90xx

closed 2020-04-16 by Ntemis