libretro/Lakka-LibreELEC #937: lakka-update fails (hosts are down)

raphaelyancey, 2020-04-19:
## Which version of Lakka are you using?

2.3.2

## What system hardware are you using?

Raspberry Pi 4

## What did you do?

```bash
lakka-update
```

## What did you expect to happen?

Lakka updates itself.

## What happened instead?

```bash
Lakka:~ # lakka-update
wget: error getting response: Connection reset by peer
:: Downloading upgrade
Connecting to le.builds.lakka.tv (195.201.56.250:80)
Connecting to le-builds.lakka.tv (104.24.109.116:443)
wget: error getting response: Connection reset by peer
:: Done, you can now reboot
```

Tried also to from a VPS of mine, same result.

awesomecoder3000, 2020-04-23:
Nightly is working

gsx750fw, 2020-04-30:
I have the same error and menu updater also doesn't seem to be working. Is there a way to see when it was last updated?

awesomecoder3000, 2020-04-30:
I Don’t know, I decided to switch to RetroPie

awesomecoder3000, 2020-05-03:
The IP addresses return 404, but I don’t know how to override the IP and set a new destination

Chlorus, 2020-05-08:
Same here - looks like it's trying to find that .index file on the build site and failing. Additionally, the script doesn't actually check wget's return code, so it'll just print the ":: Done, you can now reboot" message regardless.

awesomecoder3000, 2020-05-08:
I think the issue is that the link IPs are broken

awesomecoder3000, 2020-05-08:
But the links are not

gouchi, 2020-05-08:
I can't reproduce the issue on nightly using Generic x86_64

```
bash -x /usr/bin/lakka-update 
+ source /etc/os-release
+ NAME=Lakka
+ VERSION=devel-20200508152208-43653b3
+ ID=libreelec
+ VERSION_ID=2.9
+ PRETTY_NAME='Lakka (community): devel-20200508152208-43653b3'
+ HOME_URL=https://libreelec.tv
+ BUG_REPORT_URL=https://github.com/libretro/Lakka-LibreELEC
+ BUILD_ID=43653b3123e7db91dc3b9d64ee3e47c01ebdc49a
+ OPENELEC_ARCH=Generic.x86_64
+ LIBREELEC_ARCH=Generic.x86_64
+ LIBREELEC_BUILD=community
+ LIBREELEC_PROJECT=Generic
+ MIRROR=http://le.builds.lakka.tv
+ wget http://le.builds.lakka.tv/Generic.x86_64/.index -q -O -
+ head -1
+ FILE=Lakka-Generic.x86_64-2.3.2.img.gz
+ URL=http://le.builds.lakka.tv/Generic.x86_64/Lakka-Generic.x86_64-2.3.2.img.gz
+ rm -rf '/storage/.update/*'
+ echo ':: Downloading upgrade'
:: Downloading upgrade
+ wget -P /storage/.update/ http://le.builds.lakka.tv/Generic.x86_64/Lakka-Generic.x86_64-2.3.2.img.gz
Connecting to le.builds.lakka.tv (195.201.56.250:80)
Connecting to le-builds.lakka.tv (104.24.109.116:443)
saving to '/storage/.update/Lakka-Generic.x86_64-2.3.2.img.gz'
Lakka-Generic.x86_64  16% |*******************************    
```

raphaelyancey, 2020-05-08:
@gouchi It seems to fail on RPi (ARM) platform.

```bash
Lakka:~ # ping le.builds.lakka.tv
PING le.builds.lakka.tv (195.201.56.250): 56 data bytes
64 bytes from 195.201.56.250: seq=0 ttl=55 time=14.776 ms
64 bytes from 195.201.56.250: seq=1 ttl=55 time=14.642 ms
64 bytes from 195.201.56.250: seq=2 ttl=55 time=14.637 ms
^C
--- le.builds.lakka.tv ping statistics ---
3 packets transmitted, 3 packets received, 0% packet loss
round-trip min/avg/max = 14.637/14.685/14.776 ms
Lakka:~ # bash -x /usr/bin/lakka-update
+ source /etc/os-release
+ NAME=Lakka
+ VERSION=2.3.2
+ ID=libreelec
+ VERSION_ID=2.3
+ PRETTY_NAME=Lakka (official): 2.3.2
+ HOME_URL=https://libreelec.tv
+ BUG_REPORT_URL=https://github.com/libretro/Lakka-LibreELEC
+ BUILD_ID=bf8721be07755a8ab653a9b0941d6844651f9014
+ OPENELEC_ARCH=RPi2.RPi4.arm
+ LIBREELEC_ARCH=RPi2.RPi4.arm
+ LIBREELEC_BUILD=official
+ MIRROR=http://le.builds.lakka.tv
+ wget http://le.builds.lakka.tv/RPi2.RPi4.arm/.index -q -O -
+ head -1
wget: error getting response: Connection reset by peer
+ FILE=
+ URL=http://le.builds.lakka.tv/RPi2.RPi4.arm/
+ rm -rf /storage/.update/*
+ echo :: Downloading upgrade
:: Downloading upgrade
+ wget -P /storage/.update/ http://le.builds.lakka.tv/RPi2.RPi4.arm/
Connecting to le.builds.lakka.tv (195.201.56.250:80)
Connecting to le-builds.lakka.tv (104.24.108.116:443)
wget: error getting response: Connection reset by peer
+ [ -z ~/.update/ ]
+ echo :: Done, you can now reboot
:: Done, you can now reboot
```

That's weird because I can `wget` from my laptop without error. Seems to be a RPi specific bug.

Chlorus, 2020-05-08:
Could it be because the RPi image doesn't use stock wget? 
```wget --version
wget: unrecognized option '--version'
BusyBox v1.25.1 (2020-01-20 19:21:47 CET) multi-call binary.```

raphaelyancey, 2020-05-08:
That rings a bell, as there is a 301 redirection from http to https. Maybe this version isn't able to follow redirections?

Chlorus, 2020-05-08:
Saw this when looking up info on how busybox wget handles redirects to https:
https://busybox.net/
> 15 July 2018 -- BusyBox 1.29.1 (stable)
> BusyBox 1.29.1. (git)
> 
> Bug fix release. 1.29.1 has fixes for wget (http->https redirect) and sendmail (angle bracket parsing).****

gouchi, 2020-05-11:
Strange I could make it works on Tinkerboard 

```
+ source /etc/os-release
+ NAME=Lakka
+ VERSION=devel-20200511143630-43653b3
+ ID=libreelec
+ VERSION_ID=2.9
+ PRETTY_NAME='Lakka (community): devel-20200511143630-43653b3'
+ HOME_URL=https://libreelec.tv
+ BUG_REPORT_URL=https://github.com/libretro/Lakka-LibreELEC
+ BUILD_ID=43653b3123e7db91dc3b9d64ee3e47c01ebdc49a
+ OPENELEC_ARCH=TinkerBoard.arm
+ LIBREELEC_ARCH=TinkerBoard.arm
+ LIBREELEC_BUILD=community
+ LIBREELEC_PROJECT=Rockchip
+ LIBREELEC_DEVICE=TinkerBoard
+ MIRROR=http://le.builds.lakka.tv
+ wget http://le.builds.lakka.tv/Rockchip.TinkerBoard.arm/.index -q -O -
+ head -1
+ FILE=Lakka-Rockchip.TinkerBoard.arm-2.3.1.img.gz
+ URL=http://le.builds.lakka.tv/Rockchip.TinkerBoard.arm/Lakka-Rockchip.TinkerBoard.arm-2.3.1.img.gz
+ rm -rf /storage/.update/Lakka-Rockchip.TinkerBoard.arm-2.3.1.img.gz
+ echo ':: Downloading upgrade'
:: Downloading upgrade
+ wget -P /storage/.update/ http://le.builds.lakka.tv/Rockchip.TinkerBoard.arm/Lakka-Rockchip.TinkerBoard.arm-2.3.1.img.gz
Connecting to le.builds.lakka.tv (195.201.56.250:80)
Connecting to le-builds.lakka.tv (104.24.109.116:443)
saving to '/storage/.update/Lakka-Rockchip.TinkerBoard.arm-2.3.1.img.gz'
Lakka-Rockchip.Tinke   9% |******************   
```

Yqnn, 2020-05-19:
Same issue on Lakka 2.3.2 x64:
```
Lakka:~ # wget https://le-builds.lakka.tv/Generic.x86_64/.index
Connecting to le-builds.lakka.tv (104.24.109.116:443)
wget: error getting response: Connection reset by peer
```

I thought it was linked to TLS version, as ```le-builds.lakka.tv``` accepts only TLS1.2 or above, but it's not.
 ```github.com``` is also accepting only TLS1.2, and works well:
```
Lakka:~ # wget https://github.com
Connecting to github.com (140.82.118.4:443)
index.html           100% |************|   127k  0:00:00 ETA
```

ToKe79, 2020-05-19:
the latest nightly builds are based on LibreELEC 9.2, so there is different/newer version of busybox. However there is bug in the update script - the test should be `-f`, not `-z`.

Yqnn, 2020-05-19:
It seems this bug was specific to BusyBox 1.25.1.
BusyBox 1.26.0 is not impacted.

On freshly compiled BusyBox versions:
```
$ ./busybox-1.25.1/busybox wget https://le-builds.lakka.tv/Generic.x86_64/.index
Connecting to le-builds.lakka.tv (104.24.109.116:443)
wget: error getting response: Connection reset by peer
$ ./busybox-1.26.0/busybox wget https://le-builds.lakka.tv/Generic.x86_64/.index
Connecting to le-builds.lakka.tv (104.24.109.116:443)
.index               100% |**********|    34   0:00:00 ETA
```

toomuchnoise, 2020-05-29:
Same error on an old version of Lakka - 2.1 (Generic.x86_64).
```
Lakka:/ # bash -x /usr/bin/lakka-update
+ source /etc/os-release
+ NAME=Lakka
+ VERSION=2.1
+ ID=libreelec
+ VERSION_ID=2.1
+ PRETTY_NAME=Lakka (official): 2.1
+ HOME_URL=https://libreelec.tv
+ BUG_REPORT_URL=https://github.com/libretro/Lakka-LibreELEC
+ BUILD_ID=e1cba9bab3d7bbdcdc22292a8ce283646e6acc20
+ OPENELEC_ARCH=Generic.x86_64
+ LIBREELEC_ARCH=Generic.x86_64
+ LIBREELEC_BUILD=official
+ MIRROR=http://le.builds.lakka.tv
+ wget http://le.builds.lakka.tv/Generic.x86_64/.index -q -O -
+ head -1
wget: error getting response: Connection reset by peer
+ FILE=
+ URL=http://le.builds.lakka.tv/Generic.x86_64/
+ rm -rf /storage/.update/*
+ echo :: Downloading upgrade
:: Downloading upgrade
+ wget -P /storage/.update/ http://le.builds.lakka.tv/Generic.x86_64/
Connecting to le.builds.lakka.tv (195.201.56.250:80)
Connecting to le-builds.lakka.tv (172.67.197.143:443)
wget: error getting response: Connection reset by peer
+ [ -z ~/.update/ ]
+ echo :: Done, you can now reboot
:: Done, you can now reboot
```

igorsantos07, 2020-06-27:
Another one in a rPI here. Any solution or workaround found? :)

fvoisard2, 2020-06-29:
I get the same error, I am trying to fix the http to https, but I can't figure out how to remount the filesystem for /usr/bin so that i can rewrite the file.  this page is as close I found to anyone with the same problem

clevenger23, 2020-07-11:
any solution yet? also how do i update Flycast?

Ntemis, 2020-08-11:
lakka-update is not supported. drop the updated release into the .update folder and reboot

closed 2020-08-11 by Ntemis