libretro/Lakka-LibreELEC #939: Travado na logo do Lakka ao dar boot

NedcloarBR, 2020-04-22:
estou com um projeto de transformar meu Netbook antigo em um Video game retro usando o Lakka porém após instalar o Lakka no HD usando o USB Pendrive Flash ao dar Boot o sistema se encontra travado na logo do Lakka e não acontece mais nada e pelo Pendrive ele não entra em modo Live somente faz a checagem do hardware e reinicia e fica em loop.

Especificações do Netbook:
CPU: AMD A4-1200 1GHz
RAM: 2GB
GPU: Integrada Radeon HD Graphics

brenodantas10, 2020-05-25:
Não sei se tu já resolveu o problema e não sei se o que vou falar é o seu caso, mas até onde sei o lakka não encontra automaticamente uma tela.

Tipo, quando instalei no meu note, só aparecia na tela principal dele, não passava pro hdmi até mudar nas configurações. Quando mudei, e dei reboot sem hdmi conectado, ele ficava apenas com a bootlogo do lakka, que nem no seu caso... mas as vezes não quer dizer que não carregou, pode significar só que ele está tenta do usar outra tela.

Se for este o caso, tem que mudar as confivurações do lakka, pra mudar a tela principal

gouchi, 2020-05-26:
Can you try to boot with `live ssh textmode` and give some log at least  `dmesg`, `lspci -nnk | grep -A3 VGA` and `LIBGL_DEBUG=verbose retroarch --menu --verbose >> log.txt 2>&1`

Pode tentar arrancar com  `live ssh textmode` e dar algum log ?

obrigado

closed 2020-08-18 by Ntemis

mbmoreira84, 2021-01-16:
Estou com o mesmo problema, fica dando reboot na iniciação. Fiz a dica do hdmi, mas não funcionou.

Ntemis, 2021-01-16:
No coprendo

Reberth, 2021-08-21:
> Estou com o mesmo problema, fica dando reboot na iniciação. Fiz a dica do hdmi, mas não funcionou.

> Estou com o mesmo problema, fica dando reboot na iniciação. Fiz a dica do hdmi, mas não funcionou.

Olá amigos!
Estive passando pelo mesmo problema do @NedcloarBR e lendo a resposta ao problema oferecida pelo @brenodantas10 .
No meu caso tive que remover a placa de video, para voltar a funcionar corretamente o Lakka v3.3.

Todos que tiver este problema; tente primeiro colocar o monitor na saida VGA da placa de video, mas se ainda assim não funcionar. Remova a placa de video e tente usando o monitor na saida VGA da placa mãe.
OBS: Funcionou para mim só a pos remover a placa de video e usar na VGA da placa mãe em 21/08/2021.

Reberth, 2021-08-21:
Linux Lakka v3.3 travado na inicialização com a FLOR congelado.

Todos que tiver este problema; tente primeiro colocar o monitor na saida VGA da placa de video, mas se ainda assim não funcionar. Remova a placa de video e tente usando o monitor na saida VGA da placa mãe.
OBS: Funcionou para mim só a pos remover a placa de video e usar na VGA da placa mãe em 21/08/2021.