libretro/Lakka-LibreELEC #940: Lakka does not work on Intel nuc10i3

jtesser, 2020-04-25:
If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
2.3.2
- What system hardware are you using?
Intel NUC 10i3
- What did you do?
Graphics driver not loading. Saying Couldn't Load DRM ....
- What did you expect to happen?
Graphics driver to load. 
- What happened instead?
It didn't load 

This is a newer NUC unit but I would expect Lakka to support it.

Ntemis, 2020-04-25:
2.3.2 is using an old mesa stack and since your gpu is fairly new you should try to load a nightly image

jtesser, 2020-04-25:
I tried to find them. I would be happy to try.  Could you point me to it?

jtesser, 2020-04-25:
this is the newest I found. AM I missing something? 

https://nightly.builds.lakka.tv/2020-02-13/Generic.i386/

Ntemis, 2020-04-25:
No you are not, does that image boot ok?

jtesser, 2020-04-25:
no at that nightly comes up and to the text loader to install or enter live mode and when I try to goto live mode it just restarts

jtesser, 2020-04-25:
fwiw also tried this build https://nightly.builds.lakka.tv/2020-01-24/Generic.x86_64/

and it fails also with the DRM error 

I am happy to help test the a nightly but the only one I see to test is the one I referenced earlier 
https://nightly.builds.lakka.tv/2020-02-13/Generic.i386/

which just reboots after the text to live boot or install

Ntemis, 2020-08-11:
@jtesser please test these https://nightly.builds.lakka.tv/2020-08-10/Generic.x86_64/
and close issue if ok

closed 2020-08-18 by Ntemis