libretro/Lakka-LibreELEC #942: Please update docs @ http://www.lakka.tv/doc/Arcade

loremari, 2020-04-27:
I believe the documentation at http://www.lakka.tv/doc/Arcade/ is out of date.

It states:

> Lakka ships two emulators able to launch arcade ROMs: MAME2003 and FBA
> In Lakka, we recommend you to use FBA with an FBA v0.2.97.38 split ROM set. RetroArch doesn’t support merged sets anyway. But if you really need to run more games, choose MAME2003 with a MAME0.78 splitted ROM set

But my fresh install of Lakka (build date: Jan 20 2020) only has the following arcade cores:

- FinalBurn Neo
- MAME 2003-Plus
- MAME 2010

As a new Lakka user, I understand that these three core versions require different ROM versions from the ones stated in the Lakka docs - yet I can't find the info anywhere (tried posting  on the forum and on the Lakka subreddit but got no response yet).

Ntemis, 2020-04-27:
can you take over things like this?
every contribution counts, let us know

loremari, 2020-04-27:
Hello Ntemis, as I said I am a new user to Lakka, and I add now to emulation in general... I'm very far from being knowledgeable about the subject

Ntemis, 2020-04-27:
You can read and write English just fine though 😉

RobLoach, 2020-04-27:
https://github.com/libretro/Lakka-LibreELEC/wiki/Arcade

loremari, 2020-04-29:
So far I managed to understand (mainly from libretro docs, RetroPie docs, and wild Google searches - take everything with the benefit of doubt):

- MAME 2003-Plus requires romset called "MAME 2003-Plus", not versioned (originally a fork of MAME 0.78 with fixes and support for additional ROMs backported in)
- MAME 2010 requires romset version "MAME 0.139" (also called "MAME 2010")
- FinalBurn Neo requires romset version "FinalBurn v0.2.97.44" but this means little, as the emulator is in active development and the romset changes without being versioned. I don't know if this is still true for the FBNeo version shipped with Lakka (I don't think the FBNeo project does versions yet, only snapshots?) The recommendation seems to be to build the romset manually using the CLRMamePro tool (which looks pretty scary to me)

Based on the above, I went with MAME2010 since I installed Lakka on a 2012 laptop and don't plan to run newish games - performance is great and all the games I tested run without any problem.

Also: the doc as-is recommends "split" romsets - I couldn't find those anywhere for the above romset versions so I went with a "non-merged" set - would it be worth to change the recommendation?

I have to say that the lack of up-to-date docs is a high barrier to entry for new users like me - and I have worked in IT for 20 years now. The software is great anyway, and I appreciate it a lot, keep up the good work!

loremari, 2020-05-01:
> Based on the above, I went with MAME2010

Update: I was hit by [mame2010-libretro bug 119](https://github.com/libretro/mame2010-libretro/issues/119) where buttons A and R1 don't work for player 2. This is a no-no for me since it means my kid can't play Street Fighter II with his friends. I switched to MAME2003-Plus and the problem was solved.

closed 2020-08-18 by Ntemis

loremari, 2020-09-30:
This issue was closed with no change whatsoever to the documentation, which is still obsolete and misleading?

Ntemis, 2020-09-30:
It is closed as no one is interested in diving into this atm and there are plans to merged the  wiki with retroarch