libretro/Lakka-LibreELEC #943: Compilation of Lakka 2.3.2 for Odroid C2 FAIL

aeloya, 2020-04-29:
Hi!
I have try to compile Lakka 2.3.2 on my Ubuntu 18.04.4  
$ DISTRO=Lakka PROJECT=Odroid_C2 ARCH=aarch64 make image
and recieve this error
...
  BUILD    flycast (target)
make[1]: Entering directory '/home/src/Lakka-LibreELEC/build.Lakka-Odroid_C2.aarch64-2.3-devel/flycast-a37ee3c'
...
/home/src/Lakka-LibreELEC/build.Lakka-Odroid_C2.aarch64-2.3-devel/toolchain/bin/aarch64-libreelec-linux-gnueabi-g++  -I./core/libretro -I./core/ -I./core/deps -I./core/deps/picotcp/include -I./core/deps/picotcp/modules -I./core/libretro-common/include -I./core/deps/vixl -I./core/deps/khronos -I./core/deps/glslang -I./core/deps/flac/include -mcpu=cortex-a53+crc+fp+simd -mabi=lp64 -Wno-psabi -fomit-frame-pointer -Wall -pipe -O2   -DTARGET_NO_OPENMP -D__LIBRETRO__  -DHAVE_GLSYM_PRIVATE -DNO_VERIFY -DNDEBUG -flto -DRELEASE -DHAVE_GL3 -funroll-loops -DCORE -DHAVE_TEXUPSCALE -DHAVE_OPENGL -DHAVE_VULKAN -DTARGET_NO_JIT -DFLAC__HAS_OGG=0 -DFLAC__NO_DLL -DHAVE_LROUND -DHAVE_STDINT_H -DHAVE_STDLIB_H -DHAVE_SYS_PARAM_H -D_7ZIP_ST -DUSE_FLAC -DUSE_LZMA -O3 -c -fno-strict-aliasing -ffast-math -fomit-frame-pointer -fPIC  -mcpu=cortex-a53+crc+fp+simd -mabi=lp64 -Wno-psabi -fomit-frame-pointer -Wall -pipe -O2  -DGIT_VERSION=\"" a37ee3c8"\"  -DTARGET_NO_OPENMP -std=c++11 -D__LIBRETRO__  -DHAVE_GLSYM_PRIVATE -DNO_VERIFY -DNDEBUG -flto -DRELEASE -DHAVE_GL3 -funroll-loops -DCORE -DHAVE_TEXUPSCALE -DHAVE_OPENGL -DHAVE_VULKAN -DTARGET_NO_JIT -DFLAC__HAS_OGG=0 -DFLAC__NO_DLL -DHAVE_LROUND -DHAVE_STDINT_H -DHAVE_STDLIB_H -DHAVE_SYS_PARAM_H -D_7ZIP_ST -DUSE_FLAC -DUSE_LZMA -fno-rtti -fpermissive -fno-operator-names -fPIC core/deps/chdpsr/cdipsr.cpp -o core/deps/chdpsr/cdipsr.o
In file included from ./core/libretro-common/include/glsm/glsm.h:30,
                 from core/rend/gles/gles.h:6,
                 from core/serialize.cpp:25:
./core/libretro-common/include/glsym/rglgen_headers.h:72:10: fatal error: GL/gl.h: No such file or directory
   72 | #include <GL/gl.h>
      |          ^~~~~~~~~
compilation terminated.
make[1]: *** [Makefile:1083: core/serialize.o] Error 1

natinusala, 2020-05-01:
aarch64 is not supported, please try arm instead

aeloya, 2020-05-02:
Thank you for response!
When try DISTRO=Lakka PROJECT=Odroid_C2 ARCH=arm make image
I recieve 
...
GIT VERSION sameboy - 54f41cb
fatal: ambiguous argument '54f41cb': unknown revision or path not in the working tree.
Use '--' to separate paths from revisions, like this:
'git <command> [<revision>...] -- [<file>...]'
Makefile:29: recipe for target 'image' failed
make: *** [image] Error 128

by-gnome, 2020-05-28:
> fatal: ambiguous argument '54f41cb': unknown revision or path not in the working tree.

Branch with commit 54f41cb which should be after bd9ac20 is not exist.
Try c9e547c.

aeloya, 2020-05-31:
Today I receive new error

Connecting to rpm5.org (rpm5.org)|91.195.240.94|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Saving to: 'sources/popt/popt-1.16.tar.gz'

sources/popt/popt-1.16.tar.gz                   [ <=>                                                                                     ]  18.08K  --.-KB/s    in 0.05s   

2020-05-31 20:05:30 (375 KB/s) - 'sources/popt/popt-1.16.tar.gz' saved [18512]

INFO Calculated checksum is: 9299f31b58f7fab82cd0ee11c9f05a60c31904e4ff47fae873ddb287cb442577

  UNPACK   popt

gzip: stdin: not in gzip format
tar: Child returned status 1
tar: Error is not recoverable: exiting now
Makefile:29: recipe for target 'image' failed
make: *** [image] Error 2

rpm5.org looks unavailable. 

What can I do now?

by-gnome, 2020-06-06:
Use http://sources.lakka.tv/popt-1.16.tar.gz

Ntemis, 2020-08-11:
2.3.x is eol closing

closed 2020-08-11 by Ntemis

rodneyrod, 2021-05-25:
Just posting, no images have been listed for Odroid C2 since this post, if it has been dropped could we get an official post regarding this?

Ntemis, 2021-05-25:
If libreelec doesn't support it usually we can't do anything about it

aeloya, 2021-05-25:
http://www.lakka.tv/get/linux/amlogic/ you can find new image here

rodneyrod, 2021-05-25:
Thank you beautiful people this is perfect 😄