libretro/Lakka-LibreELEC #948: Unable to update through web interface, `lakka-update`, or `.update` folder

nfriend, 2020-05-10:
### Which version of Lakka are you using?

```
Lakka (unofficial) Version: devel-20161221183017-r21729-g274b40f
Lakka git: 274b40f7a5f7a880309a4fe8765e69ac8d8153e1
```

### What system hardware are you using?

An old Dell laptop (~10 years old) - a Dell Studio 1747.

### What did you do?

I tried to update Lakka using the following methods:

1. Ran `lakka-update` at the command line. This is the output I got:

   ```bash
   Lakka:~/.update # lakka-update
   wget: server returned error: HTTP/1.1 404 Not Found
   :: Downloading upgrade
   Connecting to sources.lakka.tv (104.24.108.116:80)
   wget: server returned error: HTTP/1.1 404 Not Found
   :: Done, you can now reboot
   Lakka:~/.update #
   ```
1. Tried using the online updater (**Online Updater > Update Lakka**). This resulted in a `Task failed: Downloading '.index'` error message in the UI after about a half a second.

1. Downloaded `Lakka-Generic.x86_64-2.3.2.img.gz` manually, dropped this in the `/storage/.update` directory, and ran `reboot`. Nothing happened from what I can tell.  I did not see any message about the upgrade process [as mentioned here](https://www.lakka.tv/doc/Upgrading-Lakka/#manual-updates), and the file still exists in the `.update` directory.
   1. To make sure this wasn't a permissions issue, I `chmod 777`'d both the file and the `.update` directory, and ran `reboot` after each. This didn't seem to work either.

### What did you expect to happen?

At least one of these methods should have worked to update Lakka.

### What happened instead?

(See explanations above.)

nfriend, 2020-05-10:
The `lakka-update` and Online Updater failures may be related to https://github.com/libretro/Lakka-LibreELEC/issues/937 🤔

nfriend, 2020-05-11:
Update: Figured it out with the help of [this post](https://forums.libretro.com/t/why-my-lakka-dont-update-autolly/6609/2)!

I dropped the `.tar` version of the update file into the `.update` directory (`Lakka-Generic.x86_64-2.3.2.tar` instead of `Lakka-Generic.x86_64-2.3.2.img.gz`) and rebooted.

I'll leave this open for now because I think it would be worth updating [this page](https://www.lakka.tv/doc/Upgrading-Lakka/#manual-updates) to mention that the `.tar` file should be used instead of the `.img.gz`.

Ntemis, 2020-08-11:
img or tar wont matter when updating

closed 2020-08-11 by Ntemis

war59312, 2022-02-19:
If you are on http://le.builds.lakka.tv/RPi2.RPi4.arm/Lakka-RPi2.RPi4.arm-2.3.2.img.gz from https://www.lakka.tv/get/linux/rpi4/

It fails to update even using https://le-builds.lakka.tv/RPi4.arm/Lakka-RPi4.arm-3.6.img.gz

You get:

Lakka-RPi4.arm-36.img.gz is not compatible with RPi2.RPi4.arm hardware - update cancelled

Current system: RPi2.RPi4.arm
Update system: Rpi4.arm

Create /storage/.update/.nocompat to disable compatibility checks and risk a non-booting system.

ToKe79, 2022-02-21:
That URL is really garbage, but nothing currently links there from https://www.lakka.tv/get/linux/ - so probably old bookmark, eh?
Anyway, as the "legacy" images were moved to a sub-folder on the download server long time ago, a download attempt should result into a 404 error (file not found).

war59312, 2022-02-21:
True enough, and yes old bookmark.