libretro/Lakka-LibreELEC #952: Wifi Not working doesn't detect interface

MattSessions, 2020-05-18:
Hi there, Love the project.

I have a Samsung laptop N510 (atom based PC) running the latest version of LAKKA.

The wifi interface in the laptop is a realtek RTL8192E ... could someone build a driver for it?

Ethernet works fine.

Thanks,

Matt

gouchi, 2020-05-19:
Hi,

Please provide output of `dmesg` and `lspci -n  | grep -E '07aa|10ec' `

Thank you.

MattSessions, 2020-05-26:
Hi. Thanks for taking a look at this for me. 

lspci:
03:00.0 0280: 10ec:8192 (rev 01)

dmesg:
[    0.000000] Linux version 4.11.12-rt14 (natinusala@helsinki) (gcc version 7.3.0 (GCC) ) #1 SMP PREEMPT RT Fri Sep 13 03:32:59 CEST 2019
[    0.000000] x86/fpu: x87 FPU will use FXSAVE
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009dbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009dc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000dc000-0x00000000000dffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e4000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000006feaffff] usable
[    0.000000] BIOS-e820: [mem 0x000000006feb0000-0x000000006febbfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000006febc000-0x000000006fec0fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000006fec1000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec0ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fff00000-0x00000000ffffffff] reserved
[    0.000000] Notice: NX (Execute Disable) protection missing in CPU!
[    0.000000] SMBIOS 2.5 present.
[    0.000000] DMI: SAMSUNG ELECTRONICS CO., LTD. N510                       /N510                       , BIOS 03MU.M009.20100125.LDG 01/25/2010
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x6feb0 max_arch_pfn = 0x100000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-BFFFF uncachable
[    0.000000]   C0000-CFFFF write-protect
[    0.000000]   D0000-D7FFF uncachable
[    0.000000]   D8000-EFFFF write-through
[    0.000000]   F0000-FFFFF write-back
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 000000000 mask 080000000 write-back
[    0.000000]   1 base 07FF00000 mask 0FFF00000 uncachable
[    0.000000]   2 disabled
[    0.000000]   3 disabled
[    0.000000]   4 disabled
[    0.000000]   5 disabled
[    0.000000]   6 disabled
[    0.000000]   7 disabled
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WC  UC- WT  
[    0.000000] found SMP MP-table at [mem 0x000f6f70-0x000f6f7f] mapped at [c00f6f70]
[    0.000000] Scanning 1 areas for low memory corruption
[    0.000000] initial memory mapped: [mem 0x00000000-0x023fffff]
[    0.000000] Base memory trampoline at [c0099000] 99000 size 16384
[    0.000000] BRK [0x02194000, 0x02194fff] PGTABLE
[    0.000000] ACPI: Early table checksum verification disabled
[    0.000000] ACPI: RSDP 0x00000000000F6ED0 000024 (v02 PTLTD )
[    0.000000] ACPI: XSDT 0x000000006FEB35F4 00005C (v01 SECCSD LH43STAR 06040000  LTP 00000000)
[    0.000000] ACPI: FACP 0x000000006FEBBC88 0000F4 (v03 NVIDIA MCP79    06040000 PTL_ 000F4240)
[    0.000000] ACPI: DSDT 0x000000006FEB4A39 0071DB (v01 NVIDIA MCP79    06040000 MSFT 03000001)
[    0.000000] ACPI: FACS 0x000000006FEC0FC0 000040
[    0.000000] ACPI: FACS 0x000000006FEC0FC0 000040
[    0.000000] ACPI: SLIC 0x000000006FEBBD7C 000176 (v01 SECCSD LH43STAR 06040000  LTP 00000000)
[    0.000000] ACPI: MCFG 0x000000006FEBBEF2 00003C (v01 PTLTD    MCFG   06040000  LTP 00000000)
[    0.000000] ACPI: HPET 0x000000006FEBBF2E 000038 (v01 PTLTD  HPETTBL  06040000  LTP 00000001)
[    0.000000] ACPI: APIC 0x000000006FEBBF66 000072 (v01 PTLTD  ? APIC   06040000  LTP 00000000)
[    0.000000] ACPI: BOOT 0x000000006FEBBFD8 000028 (v01 PTLTD  $SBFTBL$ 06040000  LTP 00000001)
[    0.000000] ACPI: SSDT 0x000000006FEB3650 0013E9 (v01 PmRef  CpuPm    00003000 INTL 20060113)
[    0.000000] ACPI: Local APIC address 0xfee00000
[    0.000000] 902MB HIGHMEM available.
[    0.000000] 887MB LOWMEM available.
[    0.000000]   mapped low ram: 0 - 377fe000
[    0.000000]   low ram: 0 - 377fe000
[    0.000000] BRK [0x02195000, 0x02195fff] PGTABLE
[    0.000000] Zone ranges:
[... 661 lines cut ...]
[   17.572743] snd_hda_intel 0000:00:08.0: Disabling MSI
[   17.651792] usbcore: registered new interface driver btusb
[   18.073949] intel_powerclamp: No package C-state available
[   18.086914] intel_powerclamp: No package C-state available
[   18.297796] snd_hda_codec_realtek hdaudioC0D0: autoconfig for ALC269: line_outs=1 (0x14/0x0/0x0/0x0/0x0) type:speaker
[   18.297806] snd_hda_codec_realtek hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[   18.297812] snd_hda_codec_realtek hdaudioC0D0:    hp_outs=1 (0x15/0x0/0x0/0x0/0x0)
[   18.297816] snd_hda_codec_realtek hdaudioC0D0:    mono: mono_out=0x0
[   18.297820] snd_hda_codec_realtek hdaudioC0D0:    inputs:
[   18.297826] snd_hda_codec_realtek hdaudioC0D0:      Internal Mic=0x19
[   18.297831] snd_hda_codec_realtek hdaudioC0D0:      Mic=0x18
[   18.874014] wl: loading out-of-tree module taints kernel.
[   18.874029] wl: module license 'MIXED/Proprietary' taints kernel.
[   18.874032] Disabling lock debugging due to kernel taint
[   18.901412] input: HDA NVidia Mic as /devices/pci0000:00/0000:00:08.0/sound/card0/input14
[   18.905035] input: HDA NVidia Headphone as /devices/pci0000:00/0000:00:08.0/sound/card0/input15
[... cut ...]
[  105.219475] perf: interrupt took too long (3170 > 3168), lowering kernel.perf_event_max_sample_rate to 63000
[  136.025294] perf: interrupt took too long (3971 > 3962), lowering kernel.perf_event_max_sample_rate to 50000
[  192.676558] perf: interrupt took too long (4989 > 4963), lowering kernel.perf_event_max_sample_rate to 40000
[  350.731784] perf: interrupt took too long (6258 > 6236), lowering kernel.perf_event_max_sample_rate to 31000
[  527.406262] perf: interrupt took too long (7829 > 7822), lowering kernel.perf_event_max_sample_rate to 25000

gouchi, 2020-05-27:
Thank you for the information.

May your try [this image](https://nightly.builds.lakka.tv/members/gouchi/952/Lakka-Generic.i386-2.9-devel-20200527163556-43653b3.img.gz) ?

MattSessions, 2020-05-27:
Yes! Thanks! That has worked! Awesome is it hard to explain what you did and how you worked it out??

gouchi, 2020-06-01:
We enabled some module in the kernel.

closed 2020-08-19 by Ntemis