libretro/Lakka-LibreELEC #954: AMD Ryzen 3400G

Sephiroth91, 2020-05-21:
Hello
I have the following hardware:

Asrock AB350 Gaming ITX
Amd ryzen 3400g with Vega 11 Graphics 
Crucical m.2 ssd 
Corsair 8gb 2666mhz

Lakka: 2.3.2

I really hope it's not because of the hardware I bought it especially this week for Lakka.

I can start the installation and it works. After the reboot, the logo of lakka is permanently displayed to me and nothing happens even when I wait an hour.
When I start the live mode on the usb stick, the same happens with live ssh.

gouchi, 2020-05-21:
Please try to provide some [log](http://www.lakka.tv/doc/Troubleshooting-Lakka/) at least graphic card `lspci -nnk | grep -A3 VGA` and `dmesg` logs.

Thank you.

Sephiroth91, 2020-05-22:
I can't start the shell. I enter "live ssh" and it only shows the logo and I can't do anything anymore. I read in an older issue that the APU should not be compatible with the 2400G and the APU on the 3400G is the same.
but I have tried the following:
https://nightly.builds.lakka.tv/Lakka-LE9.2/latest/Generic.x86_64/
and it works very well
is the version ready for production with x64 ?
I don't want to wait months until I can use my new hardware for my retro project :(

gouchi, 2020-05-22:
You can try `live ssh textmode` to get the log.

The time we release a new version you can use this [version](https://nightly.builds.lakka.tv/Lakka-LE9.2/oven/Generic.x86_64/).

Kernel and mesa must be too old for your GPU on 2.3.2.

Sephiroth91, 2020-05-22:
I try live ssh textmode. I get a blackscreen and the TV says "No Signal".
I want to try this version.

Other Question: Is it possible to add a core manually ? I would like to play playstation 3 games with Lakka. https://rpcs3.net/

When you need someone for tests I will help you.

Sephiroth91, 2020-05-22:
I tested some games. I have no sound with TV (HDMI) but when I use my Headphones on the device I have sound.

gouchi, 2020-05-23:
I see so for now you should use the Lakka-LE9.2.

Unfortunately, rpcs3 is not yet available as libretro core. Please check this [upstream issue](https://github.com/RPCS3/rpcs3/issues/2304).

Ntemis, 2020-08-11:
You have to use the lakka nightlies based on master

closed 2020-08-11 by Ntemis