libretro/Lakka-LibreELEC #956: After commit Feb/17 of 4do libretro package, the TARGET_NAME is now opera

brenodantas10, 2020-05-25:
I tried to run the libretro_update.sh and then make, the result was an error saying that the 4do_libretro.so was missing. After investigation, the 4do_libretro  project name is now opera and the githut 4do-libretro page redirects to opera-libretro page.

As the MakeFile's TARGET_NAME changed to "opera" as well, the Lakka's 4do package.mk script couldn't find the correct .so file.

- **Which version of Lakka are you using?**
 Trying to build the latest commit with updated libretro packages
- **What system hardware are you using?**
 Acer Aspire E5-754 (Intel Core i7 6500U)
- **What did you do?**
 run libretro_update.sh script and then make in the root of the project folder
- **What did you expect to happen?**
 Compile the KERNEL and SYSTEM files
- **What happened instead?**
 message of error "Couldn't find 4do_libretro.so file"

twinaphex, 2020-05-25:
This is a valid report.

@natinusala You should change this when you can.

natinusala, 2020-05-26:
@gouchi I assume you already fixed that in your recent libretro update?

gouchi, 2020-05-26:
Yes indeed ;-)

closed 2020-06-23 by brenodantas10