libretro/Lakka-LibreELEC #959: make image does not pass -j setting down

zgoda, 2020-05-30:
When running 1st build on a machine with large number of cores and with poor connection, the build completely hoses network and fails many times on git clone operations. Passing usual -j param to `make image` (eg. `make -j 8 image`) makes no difference because this param is not passed down to `scripts/image mkimage`.

natinusala, 2020-05-30:
You have to use MAKEFLAGS=-j12 for it to work

natinusala, 2020-05-30:
But I think the -j flag is automatically passed to make sub processes, I 
think I read that somewhere

zgoda, 2020-05-30:
`toolchain/bin/parallel` is running with `--max-procs 100%`. `MAKEFLAGS` does not affect `parallel` invocation.

Looking at `config/multithread` I think it's `THREADCOUNT` that affects number of processes spawned by `parallel`.

zgoda, 2020-05-30:
Indeed, this is it. `THREADCOUNT` env var controls number of processes spawned by parallel during build. So if you want to limit number of processes that do a `git clone` (eg. libretro cores) this is the way.

Later spawned compilation processes controlled by make will follow `MAKEFLAGS` setting so I guess there may be some conflict there.

closed 2020-08-18 by Ntemis