libretro/Lakka-LibreELEC #960: chailove build fails because of missing dep

zgoda, 2020-05-30:
I'm trying to build master for Allwinner H3.

```
Executing (target): make 
make[1]: Entering directory '/home/jazg/work/Lakka-LibreELEC/build.Lakka-H3.arm-2.9-devel/chailove-c7f2fb3'
make[1]: *** No rule to make target 'vendor/libretro-common/utils/md5.o', needed by 'chailove_libretro.so'.  Stop.
```

Am I missing something?

zgoda, 2020-05-31:
Apparently I am missing something, building standalone package works fine if I run `make clean` in package source directory and then `DISTRO=Lakka PROJECT=Allwinner DEVICE=H3 ARCH=arm scripts/build chailove`.

RobLoach, 2020-06-02:
Hmm, was able to reproduce in the repo itself. Started to try to clean up the Makefile over at:
https://github.com/libretro/libretro-chailove/pull/389

closed 2020-08-18 by Ntemis